Ultimaker/Cura #13592: Italian language incorrect with French

afruzu, 2022-10-18:
### Application Version

5.2.0

### Platform

Windows 10

### Printer

Artillery Hornet

### Reproduction steps

upgrading fron Ultimaker 5.2.0 beta in italian language (used without problems) to ver 5.2.0 cura mixes italian and french languages!!!
Profiles correctly imported. 

### Actual results

upgrading fron Ultimaker 5.2.0 beta in italian language (used without problems) to ver 5.2.0 cura mixes italian and french languages!!!
Profiles correctly imported. 

### Expected results

Have a correct language gestion.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

nothing

afruzu, 2022-10-18:
I've tryed to install 5.2.0 from scratch in a clean Windows 10 machine and the problem is the same: installation in english is good, when you try to change language to italian at the next start of Cura you have a mixed french/italian interface!

MariMakes, 2022-10-19:
Hey @afruzu,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 💪, this definitely looks like a mistake on our side.

We are working hard to resolve this for you and other users as soon as possible.

Sorry for any inconvenience. 😞

afruzu, 2022-10-19:
Tank you, if it can be usefull to solve the problem, I've tried also the linux version and the problem is the same.
No problem using the very well done beta version!

MariMakes, 2022-10-19:
Hey @afruzu,

A new version with fixed Italian and Chinese translations should be available here:
https://github.com/Ultimaker/Cura/releases/tag/5.2.1

Thanks again for reporting! 💪

closed 2022-10-19 by MariMakes