Ultimaker/Cura #13597: Duplicates print path are different

Ergonomicmike, 2022-10-19:
### Application Version

5.2.0

### Platform

Win10 64-bit

### Printer

Ender 3 v2

### Reproduction steps

Not a bug, per se, since Duplicate parts print acceptably. I'm just looking for confirmation.

I modeled a small part. I used the Cura "Multiply Selected" to print a few of the same parts in one batch. (Using the "Print one at a time" in the Print Sequence.)

To my surprise, the print paths for the Duplicate parts are not the same!

[In this video](https://streamable.com/ukgw6y), you can see the first part starts with the print path defining the hole in the part. That's what I expect.

But then when it prints the next part, the print head starts with the outside of the part first. Not what I expect.

### Actual results

Print paths are different for Duplicate parts.

### Expected results

I expected them to print exactly the same. (Optimizing the path for the part.)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

None

Ergonomicmike, 2022-10-19:
Thanks! 

Should I close this Issue then since there's a ticket for it? (You may close if you wish.)

MariMakes, 2022-10-19:
We can leave it open until it is actually resolved. 

Others can find that this is something we are working on, and we can update you if we plan to release a solution so you can let us know if it's the type of solution you are looking for 👍

Ergonomicmike, 2023-07-24:
I suppose that I'm bumping this, since it's still not been fixed in Cura 5.4.0.

I thought that multiplying objects would be an easy matter of simply sending the same g-code to the printer for each object with just a different starting x,y coordinate.

 But judging from the bizarre (not singularly straight) travel moves between objects. there apparently are some needed games with coordinates to make prints work at all?

Here I "needed" (err.. wanted) the seam on the far right side of the plug. And, as you can see, on the initial object, that's where it is. But then the seam wanders from multiplied copy to copy.

[image]

Ergonomicmike, 2023-07-24:
Tnx for the quick reply. I'm already printing One at a Time. 

(I haven't tried All at Once just on principle. (There would be so much stringing as the tool head moves a bazillion times from object to object.))

Ergonomicmike, 2023-08-08:
@MariMakes

Sorry for not seeing your post sooner.  (I don't see a way to make github directly notify my Profile of activity)

No, I did not have Z-seam Relative enabled. Now that I know about it, I enabled it, and violá! - all the seams line up as designed!

So now, from an ergonomic viewpoint, I wonder if "Z Seam Relative" should be automagically enabled (or at least queried if the user wants it) whenever a user uses the "Multiply Selected" models?

Thanks for looking into this.

I'll close this issue shortly.

Ergonomicmike, 2023-08-08:
[image]

MariMakes, 2023-08-09:
Thanks for the update, I'm glad that it resolves  your issue 😊

closed 2023-08-09 by MariMakes