Ultimaker/Cura #13599: Does not open the program version 5.2

NunoFilipePereira, 2022-10-19:
### Application Version

5.2

### Platform

MacOs 10.15.7

### Printer

Flsun, biqu, Tevo, ender

### Reproduction steps

Does not open the program version 5.2

### Actual results

The latest version 5.0 says it is compatible with Mac Os 10.14. But it's not true. Since version 5.0 this system has never worked properly. Do I need to take into account things other than just installing the program?

### Expected results

Does not open the program version 5.2

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Does not open the program version 5.2

x40-Community, 2022-10-19:
Read the changelog!
Cura 5.1.1 => MacOS 10.15 is no longer supported.

NunoFilipePereira, 2022-10-19:
> Read the changelog!
> Cura 5.1.1 => MacOS 10.15 is no longer supported.
> https://github.com/Ultimaker/Cura/blob/5.2/resources/texts/change_log.txt

Then they should review the information requirements. They're wrong. but version 5.1.1 I can install. version 5.2 is not.

MariMakes, 2022-10-19:
Hey @NunoFilipePereira,

@x40-Community is correct. Unfortunately, we cannot support macOS 10.15.7 anymore, since the tools that we use to build Cura is also dropped their support for these OS versions. 

Sorry for any inconvenience. 😞 
[You can read more about it here. ](
https://support.ultimaker.com/hc/en-us/articles/360011889579-On-what-operating-systems-is-Ultimaker-Cura-available-
)

Since this is expected behavior I'll be closing this issue.

closed 2022-10-19 by MariMakes