Ultimaker/Cura #13602: Nozzle collides with support on travel move

mAd-DaWg, 2022-10-19:
### Application Version

5.2

### Platform

Linux

### Printer

Creality Ender Pro 5

### Reproduction steps

1. Slice Model,
2. Print,
3. Observe consistent collisions

### Actual results

Nozzle collides with supports on travel move

### Expected results

Nozzle should not collide with supports during travel moves

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[collisions.zip](https://github.com/Ultimaker/Cura/files/9820766/collisions.zip)

MariMakes, 2022-10-19:
Hey @mAd-DaWg,

Thanks for your report. 👍 
I don't see anything strange in the project file you shared. 

[image]
Your travel settings seem to be correct and i don't see any unexpected unretraction moves. 
You can consider increasing your Travel Avoid Distance or enable Zhop When Retracted.

Do you have a picture of how your printjob failed? 
Could it be cause by stringing in old material or warping?

mAd-DaWg, 2022-10-19:
I can do you one better, a video of the collision happening. I have crazy bed adhesion, so the print hasn't failed just yet, though ordinarily it would have failed long ago. Its hitting hard enough for me to hear the collisions from a couple rooms over. Another thing to note is that its not colliding with the part, only with the support on the other side. There are no visible deformations anywhere in the print.
https://user-images.githubusercontent.com/5232999/196711751-33c0e9a7-3bec-4cb5-b9b3-c3eae3a4aed6.mp4

mAd-DaWg, 2022-10-19:
I'd like to avoid z-hops where possible, as it induces a stringing nightmare

MariMakes, 2022-10-19:
Wow! @mAd-DaWg,

Your video is super helpful. 🤩 
I'm glad your print job hasn't failed yet. 
 
I can see the unfortunate move here: 
[image]

I'll bring your issue up with the team. 🤞

MariMakes, 2022-10-26:
We've added a ticket to the backlog with the intent to improve this. 
For internal reference CURA-9797

Thanks for the report! 👍

Prunk-Crt, 2023-10-15:
Had the same issue today. 
Cura 5.4

**Printer** 
Elegoo Neptune 4Pro

**Model** 
Printhead V2 für Neptune 2 
https://www.thingiverse.com/thing:4949488

Combing: Not on outer surface
Avoid Supports wehen traveling: Off

It is even visible in the Clicer-Preview: 
[image]

It looks like Cura does not consider the nozzle size.

bioxz, 2024-04-10:
I have a similar if not the same issue. Reporting here as this was the first search result for "cura printer colliding with support". My quite new printer without any modification (default hot end, default nozzle) collides with the "Normal" type supports, I think I have seen it with the "Tree" type support as well. This failed me a few prints in the past without me figuring out what happened (beside finding parts of the print behind my print bed), I always thought this was a bed speed or adhesion problem. But in the past two days I was able to observe a clear issue with supports.

Currently I have a model with wide supports left and right and while returning from the second support to the model the printer is colliding with the support on every layer so far. Example gcode (collisions have been observed somewhere between layers 5 and 30) and video:

[clamp.gcode.gz](https://github.com/Ultimaker/Cura/files/14934127/clamp.gcode.gz)
[Print Video](https://github.com/Ultimaker/Cura/assets/1084262/e7cc9b4a-cc91-431c-a056-62323e23339f)

**Application Version**
5.7.0

**Platform**
Linux

**Printer**
Creality Ender 3 v3 SE

bioxz, 2024-04-13:
And there goes another print. 

[iss_mount.gcode.gz](https://github.com/Ultimaker/Cura/files/14965727/iss_mount.gcode.gz)

After about two hours the nozzle suddenly kicked a support (this time I went for trees instead of the default supports) off the bed. I'm far from being an expert, but this travel on layer 208 looks suspicious and visual comparison to the print confirms that layer 208 was the last one printed on the support I found on the ground:

[image]

There is a travel right through the support, which I guess shouldn't be there like that. "Avoid Supports When Traveling" is set to the default value and is thus enabled.

I wonder what causes this. In the beginning I though I have an issue with bed leveling or something like that, but it's getting more and more clear that all my prints actually fail because of Cura placing travels through supports. Given that this report is one and a half years old and on backlog, there must be something that affects only a few users.