Ultimaker/Cura #13603: Configurable ghost model rotation

Xyem, 2022-10-19:
### Is your feature request related to a problem?

The ghost models of delta printers can be "misrotated" in Cura as there is not one correct rotation for deltas, as it is dependent on firmware and hardware configuration.

This has led to some users to report it as incorrect, it being "fixed" for them, but breaking it for other users. Comments from (what I believe are?) the Cura team indicate they want consensus on what the correction rotation is, but this isn't possible as there simply isn't one.

### Describe the solution you'd like

All model files should have a consistent orientation, such as X being parallel to the front of the machine, denoted by a user interface element such as a screen and then the rotation of the ghost should be a machine setting so they the user can align it as required.

### Describe alternatives you've considered

The current solution of "fixing" the rotation by modifying the model files is insufficient for dealing with the lack of "correct" rotation for deltas.

### Affected users and/or printers

This will primarily affect users with delta printers as this is a problem specific to their 
construction/mechanics.

### Additional information & file uploads

As neither of the two deltas I have operational match what Cura shows and *also differ to each other*, I would find this removes quite a mental burden when alignment is required e.g. during calibration prints or for orientating models for the camera.

I would be willing to put in the effort to submit it as a PR for merging, if I was assured such functionality would be accepted.

Xyem, 2022-10-19:
Ping for those who I have seen who may find this fixes the issue for them or may otherwise be interested.

@berberic2 @LuizPonzini @skoroneos @RVillani @JohanPluister @matthewbaggett

skoroneos, 2022-11-06:
Do we need a "custom" rotation when all that its need it is to get the 3mf file to the correct rotation for each printer?

Xyem, 2022-11-06:
> Do we need a "custom" rotation when all that its need it is to get the 3mf file to the correct rotation for each printer?

Yes. I explained why in my initial post. There is no "correct rotation" for delta printers. Your proposed solution has been tried and does not work due to this. See 9ffc6bc70587c5e53a43fba47f3aa1bc20eda87d for demonstration.