Ultimaker/Cura #13609: Assymetry in Bottom Layer

Ergonomicmike, 2022-10-20:
### Application Version

5.2.0

### Platform

Win 10 64-bit

### Printer

Ender 3 v2

### Reproduction steps

I printed the Calibration Cube that's part of the Calibration Shapes 2.1.5 Extension from the Marketplace. (Thank you.)

Print it.
Remove it from the Plate and look at the bottom of the cube to inspect the Bottom skin.

### Actual results

Here's a close up photo of the bottom of the cube. (I marked the "X" and the "Y" side in the photo for reference.)

[image]

As you can see, the Bottom layer is gap free to Inner wall on one half of the cube's bottom. Whereas on the opposite side of the cube - as marked by the diagonal (blue dashed line) - there are gaps.

### Expected results

I expected the bottom layer to be perfectly symmetrical. IIRC, that was the typical result with previous versions of Cura. (5.0)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

1) I used to use Coasting, and had Coasting tweaked well so that there were no gaps.
2) But now I use Linear Advance in Marlin 2.1.1, with a K=0.9. (Bowden with metal extruder, Capricorn tube.)
3) Still, I don't recall seeing this asymmetry until lately.
4) Firmware retraction.

[Calibration Cube with asymmetric gaps in bottom skin.zip](https://github.com/Ultimaker/Cura/files/9828022/Calibration.Cube.with.asymmetric.gaps.in.bottom.skin.zip)

GregValiant, 2022-10-20:
Thanks for the reporrt.
I can't duplicate that.  I sliced your 3mf file without making any changes and printed on my Ender 3 Pro (3 years old, 8 bit 1.1.5 mainboard, all-metal hot end, no linear advance, no ABL).  The PLA is some wet stuff I use for this tests.
Your slice is on the left.  Both of these were printed on the smooth side of the glass and both prints are "as smooth as glass".  The texture you see is normal in the silky silver PLA.
[image]

This one was printed on the textured side of the glass.
[image]

You recently made changes to the printer.  You might need to revisit them or recalibrate.

I don't see a bug here or a settings issue.  It appears to be a printer/calibration problem.  Other than the "retraction distance" difference (which I left at 1.5 for that print) my normal settings are nearly identical to what you have in that file.
I'm going to remove the bug label but I'll leave the report open so you can respond.

Ergonomicmike, 2022-10-20:
@GregValiant Are you Bowden or Direct Drive? 

I'm running a Slice Engineering All-metal Heat Break, So we're the same there.

GregValiant, 2022-10-20:
I've changed the hot end and the layer cooling blower (single 5015) but the rest is pretty stock and still a bowden.  I'm a decent wrench and worked in machine shops and I can tune these pretty well.
Looking at your photo I could convince myself it's under-extruding because the extrusions are still so "round" instead of smeared to the build plate.  But that "look" can also be caused if the nozzle is too high on the initial layer.  The other thing is that it appears that the initial prime of your extruder isn't kicking the filament far enough.  Coasting can cause that, but with coasting it usually happens after very long coasting moves that give the nozzle a chance to ooze and drop pressure.  That cube is pretty small and I wouldn't expect that so it's back to the extruder.  You might consider experimenting with letting Cura do the retracts and primes in the gcode.  There may be something going on in the printer when it reads the G10 and G11 commands.

Don't go making a bunch of changes at once.  That just makes troubleshooting more difficult.  A tweak here and a tweak there and you should get it printing well.

Ergonomicmike, 2022-10-20:
Tnx.

Ergonomicmike, 2022-10-24:
@GregValiant is correct. I had two problems:

1) I rechecked my mesh. it was low in the upper right corner.

2) My firmware seems to "jam up" every now and then, where I can no longer change Z-offset live. (I have to reflash occasionally.)

I changed over to using a (metal) feeler gauge and now my layers are beautiful.

closed 2022-10-24 by Ergonomicmike