Ultimaker/Cura #13610: Slicing fails with an unexpected error in 5.2.1

WojTechnical, 2022-10-20:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Creality Ender 5

### Reproduction steps

Load the project, click slice. After about 10 seconds of slicing a message is shown "Slicing Failed with an unexpected error".

### Actual results

Slicing Fails with an unexpected error. 

The error reported in the log is:

2022-10-20 10:11:59,739 - ERROR - [MainThread] CuraEngineBackend.CuraEngineBackend._onBackendQuit [964]: Backend exited abnormally with return code 3221225477!
2022-10-20 10:11:59,741 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [360]: Attempting to kill the engine process
2022-10-20 10:11:59,742 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [366]: Killing engine process
2022-10-20 10:11:59,742 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [369]: Engine process is killed. Received return code 3221225477

### Expected results

Slicing of the model should finish successfully.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9828265/cura.log)

Project file:
https://drive.google.com/file/d/1sT7HC0DsGRqs928e6NetTB7lbliNm6gT/view?usp=sharing

GregValiant, 2022-10-20:
Thanks for the report.
That model has errors and the mesh is extremely dense.
Just for a point of information there are:
480 vertices/mm²
160 triangles/mm²
At that scale (1:34,760,000) Mons Huygens would be 0.16mm tall.
Someone from the Cura team will take a look.

WojTechnical, 2022-10-20:
@GregValiant - hi! thanks for the reply.
Yes, it's an extremely detailed model, so I imagine it might be actively difficult to process(btw, 3 inch version of this model slices and prints fine, but takes few minutes to slice). 4 inch models and larger all fail to slice.

More details here:
https://www.thingiverse.com/thing:4102658

GregValiant, 2022-10-20:
I tried to slice it but this old laptop was having none of it.  I did finally get it to load in Mesh Mixer as that's where I got the vertices/triangle numbers.  The other softwares (Cura, PrusaSlicer, 3D Builder) all crashed.
The Cura team has more capable hardware (it could hardly be less) and will figure it out.

GregValiant, 2022-10-20:
I decided to try again.  MS 3D builder was able to repair the model and it sliced fine in 5.2.0.
That allows me to say "I can't duplicate the problem."
For the first test I didn't trust my computer to do all the support calculations but it sliced fairly quickly without support.
[image]

And here it is with support.  Slice time was about 2.5 minutes.  It's a 12.5 hour print with the grip infill at 10%.
[image]

WojTechnical, 2022-10-26:
@GregValiant Hi Greg, sorry for the late reply - I just wanted to say I have tested this and yes, loading the model in MS 3D Builder and re-saving it allows Cura to slice the model "some" of the time. For example I found that slicing at 0.2mm layer height works every time, but slicing at 0.15m fails sometimes(it does seem non-deterministic, not sure how that's possible).

Anyway - thank you for your time! I really appreciate your help with this :-)

GregValiant, 2022-10-26:
There was a bug in 5.1 that kept certain models from slicing and all the reports were from people running Windows.  The resolution of the model seemed to be a contributing factor.  The bug was worked on for 5.2 but seems to persist with some models.

MariMakes, 2023-02-23:
Hey @WojTechnical,

Return code 3221225477, is actually a super generic error, even though it doesn't sound generic 😅
In the Cura 5.3 Beta, we made a fix that should solve a number of these errors.
Can you download the Beta and retry your slice?
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.3.0-beta.1

If that's it still doesn't work, we'll need a little bit more information from you.
Do you have a project file for us? It contains the printer and settings we need for troubleshooting.
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the view it on GitHub link at the bottom of your email to open GitHub in the browser.

closed 2023-03-10 by github-actions[bot]