Ultimaker/Cura #13611: [macOS] stated system requirements not properly embedded in the app via Info.plist => LSMinimumSystemVersion

core-code, 2022-10-20:
### Application Version

any

### Platform

macOS

### Printer

none

### Reproduction steps

1. download macOS version (e.g. Ultimaker-Cura-5.2.1-mac.dmg )
2. try to launch it on a "too old" macOS version that is not officially supported by this release (according to your documentation here: https://support.ultimaker.com/hc/en-us/articles/360011889579 ) 

### Actual results

Ultimaker Cura launches but misbehaves on unsupported macOS versions

### Expected results

Ultimaker Cura should never launch on a macOS version not officially supported by this release. as with any other macOS app this needs to be ensured by using the appropriate `LSMinimumSystemVersion` tag in your `Info.plist` file

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

adding
```
	<key>LSMinimumSystemVersion</key>
	<string>11.0</string>
```
to your `Info.plist` file will fix this bug, but this needs to be updated any time you raise system requirements.

MariMakes, 2022-10-25:
Hey @core-code,

Welcome to the Ultimaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

I really like this suggestion. We often get questions about this. 
I'll bring it up to the team to see if they know what to do with this information.  🤞

MariMakes, 2022-10-26:
I brought it up to the team and they were super happy with your report. 😊 

We've added a ticket to the backlog with the intent to improve this. 
For internal reference CURA-9795

Thanks for the report! 👍