Ultimaker/Cura #13615: Very old bug with (not) displaying file name and parameters.

Vipcioo, 2022-10-20:
### Application Version

4.0 or less up

### Platform

Windows 10

### Printer

Not relevant

### Reproduction steps

I changed the display to a dark theme.

### Actual results

I can't see the name or parameters of the file when zoomed in.
Only when you move away from the table to darker background does all the print data correctly appear.

### Expected results

Change the colours just like a light theme where the fonts are shown against a light and dark background. 
It's impossible now.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[image]

MariMakes, 2022-10-31:
Hey @Vipcioo,

Thanks for your report 👍
Sorry, it took us a while to get back to you 😞 

I recognize your issue. We've received some internal reports on the issue. 
I asked around and we seem to have resolved it for mac users but not for Windows. 
I'll bring it up again to see if we can resolve it for widows as well. 🤞 

[image]

MariMakes, 2022-11-02:
Hey @Vipcioo 

We've added a ticket to the backlog with the intent to improve this. 
For internal reference CURA-9840

Thanks for the report! 👍

Vandresc, 2023-05-26:
Fixed for the next release!

closed 2023-05-26 by Vandresc