Ultimaker/Cura #13618: Stitch the layers using injection molded rivets

Harvie, 2022-10-21:
### Is your feature request related to a problem?

Part delaminates

### Describe the solution you'd like

Hello guys,
i have this idea to increase layer adhesion and prevent delamination.
Basicaly you create cavity which is used as injection mold to create "rivet" bonding several layers together.
This will probably need to increase temperature so the filament flows more smoothly into such mold.
What do you think?

[image]

### Describe alternatives you've considered

Viable injection depth is obviously limited, but we can use interleaved pattern like this:

[image]

It might be even possible to inject butterfly shaped rivets to increase hold.
Also even for perimeters not straight in Z direction it might be possible to get injected:

[image]

Although the injection process might require the rivet hole to be vented in some way for the air to escape:

[image]

Maybe even micro rivets (only joining two adjacent layers) might be viable, since they would eliminate need for venting and would be well suited for complex geometry while still simple to implement. But it would obviously come with reduced length of polymer chains in Z direction...

### Affected users and/or printers

FDM

GregValiant, 2022-10-22:
Thanks for the suggestion.
@nallath may have a comment.

fieldOfView, 2022-11-01:
Though I initially liked the idea, I don't think it will work very well.

In injection molding, the molds are heated to a fairly high temperature so that the injected material does not instantly solidify when it touches a wall, blocking the flow of material in small sections of the mold. In FDM printing, you want the deposited material to cool down before the next layer is deposited. In other words: the "rivet holes" you will be trying to fill will have cold walls, and it will be hard to fill them with material (especially since the walls of the holes are "ribbed" because of the layers). 

There's also an issue that when the nozzle stops over the "rivet hole", it will start to deform the print. This is already a problem for micro-pauses (~tenth of a second) caused by buffer underruns when printing via USB.

I think that printing the rivet hole solid, and pushing the nozzle down on it for half a layer height (and lifting up again before continuing) will get you the similar strength with none of the flow problems. You would need a sufficiently thick wall for that though (depending on the width of the shoulder of the nozzle).

nallath, 2022-11-01:
The rivet holes are, as far as I remember, patented. It was one of the reasons for me to come up with an alternative method (which is also patented, but by Ultimaker, so we could implement that!). Instead of leaving holes in the first place, it pushes the nozzle down to create a somewhat similar effect.

You can find more info here: https://worldwide.espacenet.com/patent/search/family/063638265/publication/EP3843976A1?q=pn%3DEP3843976A1

fieldOfView, 2022-11-01:
> it pushes the nozzle down to create a somewhat similar effect.

Heh, I though I just came up with that in my comment above ;-)