Ultimaker/Cura #13620: The "Spiral outer contour" mode, the setting does not work "Small Hole Max Length"

Mikhail2623, 2022-10-21:
### Application Version

5.2.0

### Platform

Windows 10

### Printer

CyberDragon

### Reproduction steps

Turned on the "Spiral print outer contour" mode.
Turned on the "Small Hole Max Length" mode.
Adjusted the diameter of 20mm.
Set the speed to 50%
Slicing

### Actual results

The printing speed does not change during the entire printing process.

### Expected results

In the upper part of the model, the speed should decrease by 50%
In Cura 4.13.1, this parameter works

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Cura 5.2.0.zip](https://github.com/Ultimaker/Cura/files/9840342/Cura.5.2.0.zip)

GregValiant, 2022-10-21:
Thanks for the report.
I opened the 4.13 file in 4.13 and sliced it and this is in the gcode.
;LAYER:98
M106 S184.9
;TYPE:WALL-OUTER
;MESH:?????????????????? ?????????????? ????????.STL
G1 **F342.9** X47.959 Y35.452 Z49.302 E1427.10172

Then I opened the 4.13 file in 5.2.1 and sliced it and this is in the gcode.
;LAYER:98
M106 S185.7
;TYPE:WALL-OUTER
;MESH:?????????????????? ?????????????? ????????.STL
G1 **F342.9** X47.953 Y35.466 Z49.302 E1441.76922

Then I opened the 5.2 project in 4.13 and sliced it.
;LAYER:1
M107
;MESH:?????????????????? ?????????????? ????????.STL
G0 F1050 X27.207 Y29.524 Z0.8
;TYPE:WALL-OUTER
G1 **F600** X27.617 Y29.14 Z0.804 E49.23129
That line is the only G1 speed setting line in the file.

This looks like a settings issue to me.  The 4.13 project file works the same in 4.13 and 5.2 and the speeds are adjusted.  Something is different in the 5.2 project file as it produces the same gcode (no speed changes) when opened in 4.13.
I'm inclined to remove the bug label as it does seem to be a settings problem.  I'll leave this open so you can respond here.

BTW I've never seen anyone print a 17.0mm line width using a .4 nozzle.  How do you really have the printer set up?

Mikhail2623, 2022-10-22:
Good afternoon!!
`BTW I've never seen anyone print a 17.0mm line width using a .4 nozzle. How do you really have the printer set up?`
Up to 2.0 mm is printed without difficulty, with acceptable accuracy.

`Something is different in the 5.2 project`
Thanks for the hint, there really is a problem in the print profile "PETG_BF_D0.4_0.5_1.7" Probably a profile violation occurred during updates.

I downloaded and configured the profile from the version 4.13 and the print speed began to work normally

Thanks for the hint! 
The issue is resolved.

GregValiant, 2022-10-22:
I'll go ahead and close it then.
Check your "Infill Line Width" under Quality.  Printing a .4 nozzle at 1.7 wide is one thing.  Printing it at 17.0 wide is another.

closed 2022-10-22 by GregValiant