Ultimaker/Cura #13623: Brisging sometimes doesn't bridge

Triangulix, 2022-10-22:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Custom

### Reproduction steps

Slice the attached project

### Actual results

Layer 4 is printed a normal layer.

### Expected results

The gap below layer 4 should be bridged.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Bridging.zip](https://github.com/Ultimaker/Cura/files/9844646/Bridging.zip)
[image]

MariMakes, 2022-10-25:
Hey @Triangulix,

Thanks for your report 👍

At first glance, it might be related to the Bridge Skin Support Threshold. 
The setting might be too low to actually show bridging. 

Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

Triangulix, 2022-10-25:
The atttached zip file contains the project.

MariMakes, 2022-10-26:
Hey @Triangulix,

You are correct. I must have overlooked it. 

I changed the Bridge Skin Support Threshold, and this results in the bridge being bridged. 
[image]

Does this resolve your problem? 👼

MariMakes, 2022-10-27:
If you check the layer before the layer that needs to be bridged, you can see that about 75 percent does not need any bridging because it has a layer of skin underneath it. That is higher than the default 50 percent, which is why it's treated as normal skin instead of a bridge. 
[image]

You can read more about the setting in the settings guide. 
https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/experimental/bridge_skin_support_threshold.md

Would you agree it's intended behavior and we can close the issue? 👼

Triangulix, 2022-10-27:
Okay, I understand.

closed 2022-10-27 by Triangulix