Ultimaker/Cura #13624: Ultimaker Digital Factory startup no longer creating a 6 digit code

GoguyT3D, 2022-10-22:
### Application Version

5.2.1

### Platform

Windows 11

### Printer

Ultimaker S3

### Reproduction steps

Firmware 7.1.3 on Ultimaker S3, while having numerous problems with my material colors being duplicated when uploaded to my printer via usb because sync via network doesnt work(even with firewall on pc and printer off) I decided to Reset Digital Factory and Reset all printer settings, after managing to rename my printer to the same name it was before, I also removed the printer from ultimaker digital factory on my PC. Since the reboot on my printer I have set up a local netowrk connection, and clicked on Ultimaker digital factory in settings, clicked connect and no 6 digit code showed up. Now I have no printer set up on Ultimaker Digital factory on my pc yet my printer says its connected to it, when I click on add printer I cant use any other data that a 6 digit code which my printer will no longer generate.

### Actual results

No 6 digit code appearing when I click "connect" under Digital Factory in settings

### Expected results

A six digit code should appear on my printer (especially after a Digital Factory Reset) and I should be able to see my printer on my pc web browser in ultimaker digital factory

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

No matter how hard I try I can't start fresh with Ultimaker

MariMakes, 2022-10-25:
Hey @GoguyT3D,

[I recognize your issue. ](https://community.ultimaker.com/topic/41838-there-seems-to-be-a-problem-with-syncing-my-materials-via-network)  😄 
I flagged it to the team and they are taking a look as we speak. 

Fingers crossed 🤞

GoguyT3D, 2022-10-25:
thank you

> Hey @GoguyT3D,
> 
> [I recognize your issue. ](https://community.ultimaker.com/topic/41838-there-seems-to-be-a-problem-with-syncing-my-materials-via-network) 😄 I flagged it to the team and they are taking a look as we speak.
> 
> Fingers crossed 🤞

Thank you

GoguyT3D, 2022-10-28:
Solved by Daniel De Heus (ultimaker) thank you for help

closed 2022-10-28 by GoguyT3D

Intechgreater, 2023-09-18:
> Solved by Daniel De Heus (ultimaker) thank you for help

So what was the solution? Having the same problem of the 6 digit code not being generated. Just a series of hyphens...

Intechgreater, 2023-09-18:
> > Solved by Daniel De Heus (ultimaker) thank you for help
> 
> So what was the solution? Having the same problem of the 6 digit code not being generated. Just a series of hyphens...

AH! The printer *looked* like it had a good Wifi connection, but there was no internet.