Ultimaker/Cura #13626: error -atmat_signal_pro_400_v1

prankster181, 2022-10-22:
### Application Version

5.1.0

### Platform

windows 10 64 bit

### Printer

cr-10 s5

### Reproduction steps

installed new version 5.1.0
imported the prebuilt profile for the printer.
after closing and reopening cura, a pop-up showing error -atmat_signal_pro_400_v1 wanting to delete all printers and profiles comes up.
the ender 3 v2 neo i also have is no problem. if i delete all instances of the s5 from subfolders, everything is normal.
i'm sure it has something to do with the prebuild for the s5 printer.

### Actual results

closing the popup and ignoring it seems to work.

### Expected results

5.2.1 just installed as typing this up. problem is gone. i would like to know what the issue was in case of future issues of similar type.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

don't really have anything else to add.

MariMakes, 2022-10-25:
Hey @prankster181,

Thanks for your report 👍 
I'm happy to hear your issue is resolved. 😊 

I can bring your question up to the team to see if they recognize your error. 🤞

MariMakes, 2022-10-26:
Hey @prankster181,

I brought your error up to the team.

You probably say a configuration error related to this printer
https://github.com/Ultimaker/Cura/blob/main/resources/definitions/atmat_signal_pro_400_v1.def.json

There are a number of reasons why that error shows up. 
It might have been that you have manually installed this material before, and it was now included in your latest cura version
You might have tried to delete the printer while it was still active

The next time you see that error, it might help to install the start-up optimiser. 
https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/StartOptimiser

I'll close this issue since it's resolved. 👍

closed 2022-10-26 by MariMakes