Ultimaker/Cura #13627: Allow using a cube-mapped bump map for the fuzzy feature.

poikilos, 2022-10-22:
### Is your feature request related to a problem?

Currently, to create an effect of bump mapping (displacement mapping technically), users of any slicer must create dense (up to or greater than a polygon-per-pixel) meshes.
- The realtime consequences of this are reduced performance in Cura's "Prepare" mode, editing speed in external software, and possibly other consequences.
- The non-realtime consequences models are:
  - speed of processing (including increased slicing time in Cura, and other delays in other software such as compression/decompression)
  - server issues: space for uploads (a size of at least 10 times compared to a modeled but not displaced mesh in many cases, often well over 100MB), upload time, download time, performance (cumulative load across users)
  - editing issues: self-intersecting geometry or extraneous geometry (causing the destruction of edge loops and other mesh landmarks) hinders the next designer down the line trying to improve creative commons models
  - possibly other consequences.

### Describe the solution you'd like

The solution seems to adapt the existing Fuzzy Skin feature to have an option to change the type of input from "Random" to "Bump Map". Then the detail could be added during slicing rather than during modeling, following the industry standard if slicing is considered a form of "rendering." Detail of any pattern can be faked, as long as it is shallow enough to not create new overhangs (suggested calculated maximums are below). The features would become visible if the existing fuzzy_skin ("Fuzzy Skin") checkbox is checked. The following settings would be required:
- Settings below appear if fuzzy_skin==true:
  - fuzz_type (drop-down): "Random" (default), "Bump Map"
  - fuzz_depth `*` (minimum 0; maximum .5; default is calculated: outer_wall_line_width*0.25; warn if wider; overhangs will probably be horrible if approaches outer_wall_line_width*.5 or larger but maybe increase max to outer_wall_line_width*.5 and allow it but with a yellow shade on the edit box as a warning)
  - fuzz_offset `*` (default 0 to make 50% [gray] have no effect; min -0.5 to make everything under 100% [white] push the surface inward, max 0.5 to make everything above 0% [black] push the wall outward)
  - Features below appear if both fuzzy_skin==true and fuzz_type=="Bump Map":
    - fuzzy_map_path (with browse button to choose a compatible image file)
    - fuzzy_map_mapping (drop-down): "Cube", "Cylinder" (optional)
    - fuzzy_map_repeat_h: double (horizontal repetitions around the model; minimum .001 or so in case someone is being clever with effects, but avoid divide by 0)
    - fuzzy_map_repeat_v: double (vertical repetitions along the z axis; minimum .001 or so but avoid divide by 0)

Caveats:
- `*` If the detail is of the same depth as fuzz, only mapping and file reading code is necessary (fuzz_depth and fuzz_offset can be excluded from the GUI and the values can be hard coded to the defaults. The user can adjust the image brightness in a different program to adjust the outcome). Other fuzzy code would be reused.
- Use luma (((float)R+(float)G+(float)B)/(765.0) for color images, or brightness ((double)G/255.0) for grayscale images.

### Describe alternatives you've considered

Most people who want this type of thing apply a displacement map before uploading an stl, causing it to have a huge number of polygons, causing the aforementioned delays even on an older i7 with similar benchmarks as modern ones. One outcome I've seen is hanging for over a minute just by switching to edit mode in Blender (in the worst case scenario with a very detailed mesh plus multiple boolean modifiers). More polygons seems to be the only other way to add displacement mapping.

### Affected users and/or printers

Anyone who wants the part to have non-random bumpy surface will benefit from this. Ultimaker itself and any other company or person utilizing the internet or networks for transferring models will benefit from this any time they or their customers have this use case. This problem was solved in gaming, CAD, and 3D animation pre-production quite a while ago: bump mapping (called displacement maps in terms of physically-based rendering). The same realtime rendering technology applies to the "Prepare" view of Cura, and to various programs used to model prior to opening the model in Cura.

### Additional information & file uploads

Some additional caveats seem to be out of scope but could be solved with this new iteration to the Fuzzy Skin feature or in a later iteration:
- Using normal maps would be potentially possible using the normal as a vector then applying the bump_depth as a vector length multiplier. This may only work well if bump_offset is 0 and a value of .5 (50% gray) implies the original location of the wall.
- Setting the bump_depth such that its absolute value is a number higher than .75*outer_wall_line_width: Transitioning from fuzz_depth at the wall to 0 offset to at the center of the model would allow features deeper than the outer wall, but that seems out of scope and likely to cause unpredicted overhangs unless supports are calculated after slicing. Such a feature could be considered for a future iteration if it is too much trouble.

### External Links

There are some possibly different notes that I made at the following PrusaSlicer (not Cura) issue:
- https://github.com/prusa3d/PrusaSlicer/issues/8649

For developers trying to implement this and test it, 5.2.1's "What's New" slideshow has a slide that says Cura is now easier to build. Here is the link it provides:
- https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source

Example models:
- An example of a model that would be extremely difficult to modify and maintain if using dense polygons instead of the new feature is below (my remix of Dice Tower with Fold-up Trays). I've released this model as [CC BY-SA 3.0](https://creativecommons.org/licenses/by-sa/3.0/) on Thingiverse (and attached) and it can therefore be downloaded and used for any kind of test under those terms: [Dice Tower with Fold-Up Trays, Aligned](https://www.thingiverse.com/thing:4077316)
  - Or download from GitHub:
    - [Dice_Tower_with_Fold_Up_Trays-Aligned-1.2.zip](https://github.com/Ultimaker/Cura/files/9845146/Dice_Tower_with_Fold_Up_Trays-Aligned-1.2.zip)

Example displacement maps:
-  [image] ([CC0](https://creativecommons.org/publicdomain/zero/1.0/) by https://ambientcg.com/view?id=Bricks076A)

PhilBaz, 2022-11-28:
This would be amazing!

TQM21, 2024-10-21:
I came here to suggest changes to using the fuzzy skin function to create a simple _regular_ pattern for surfaces.  Any shallow regular pattern can _greatly improve_ the visual appeal of flat surfaces by hiding/disguising the layer lines, and assist in hiding the Z-seam.

**User control of the random effect, down to 0 if desired, would be a good start.**  Allow triangle or trapezoid shape.  User control of duty cycle.  User control of rotational offset increment for each band.  User control of how many print layers per band.  Something like that.

@Poikilos bump map feature request is even better, and orders of magnitude better expressed than I ever could!  

As already said, it is _much more_ efficient to add texture in the slicer than modeled onto the object.   Hoping this comment gets his feature request more support.  Bump!