Ultimaker/Cura #13629: Keypresses are dropped when typing in new setting values with number pad.

ansonl, 2022-10-23:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

UMO+ DXU

### Reproduction steps

1. Highlight value to start entry focus on print setting that affects visibility/enable of another settings such as:
  - Top/Bottom Thickness
  - Bottom Thickness
2. Type the value you want quickly (with keyboard number pad) starting with something like 0.6. If the previous value was a something else like 0.4, the value will be replaced with 0 first and the UI will freeze for a few seconds before accepting new input. The UI will miss your next two inputs of .4 during that time and print settings will adjust show/hide/enable as if you just typed in 0 and things will be highlighted for any invalid values as a result of 0. 

### Actual results

UI misses keypresses and you have to either purposely type in number values slowly or type it all over again.

### Expected results

UI doesn't freeze and handles the full correct user input without dropping input. 

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9845548/cura.log)

MariMakes, 2022-10-25:
Hey @ansonl,

Thanks for the report 👍 
I've encountered it myself as well. 😖 

This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/13629
We have a ticket on our backlog with the intent to improve this behavior.
I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/13629

closed 2022-10-25 by MariMakes

ansonl, 2022-10-25:
@MariMakes did you mean to put another ticket number for the duplicate link?

MariMakes, 2022-10-25:
Hey @ansonl, You are right! 
By accident, I included the link to this ticket instead of the new https://github.com/Ultimaker/Cura/issues/12949. 
I've updated the comment.