Ultimaker/Cura #13632: 5.1.0 Vase Mode Extrusion to Random Point

jjz015, 2022-10-23:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Ender 3

### Reproduction steps

Attempted to print airfoil via vase mode, however there is an issue where the last bottom layer will have a move towards a random point or to infinity in either x or y direction. Changing the seam location alters which way the tool moves, and changing the wall width will also affect location. 

### Actual results

Vase mode, seam to shortest, wall width to 0.38 (w/ 0.4mm nozzle). Tool will move in some unintended direction, either to origin, random point on bed, or 

### Expected results

Toolhead should move along toolpath, no extrusion to random point

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[bugReport.zip](https://github.com/Ultimaker/Cura/files/9846265/bugReport.zip)

GregValiant, 2022-10-23:
Thanks for the report.
This looks like a duplicate of #13567 and I'll mark is as such.  There appears to be a bug in the interaction between Surface Mode, Spiralize, and Remove all Holes.  I'll close this one in favor of the previous report.  If you feel that is premature let me know and I'll re-open this one.  (You can respond here or in the other thread.)
For now, if you play with the three settings you can get this problem to go away.  

The 3mf file you included appears to be a model file rather than a Project File.  A Project File ("File | Save Project") would be very helpful trying to debug this.

That piece doesn't qualify for being printed using Spiralize.  There are different distinct areas within the model which cause long extrusion moves between the areas and results in a mess.  With Spiralize turned on take a look at the bottom layer and you will see what I mean.
This screenshot is the bottom layer and with Spiralize turned on.  It looks like separate parts.
[image]

This is layer 33 in Spiralize.  You can see all the long strings.  Those are extrusions and will be thick.
[image]

Right above the Spiralize option is "Surface Mode".  Setting it to "Surface" it looks a lot better.  This is also layer 33.
[image]

jjz015, 2022-10-23:
Ah, the other ticket does seem to be the same issue! Yes I'm fine w/ closing this ticket. I've attached the project file. I was able to get spiralize to follow the outer contours, but the surface mode does also seem to do the trick. I'll try playing around w/ the settings you've mentioned, thank you!
[Project File.zip](https://github.com/Ultimaker/Cura/files/9847716/Project.File.zip)

GregValiant, 2022-10-24:
Thanks for the file.
I see you have the line width at 0.35.  I know weight is an issue but that looks a lot better to me when sliced at 0.50 line width.  0.60 would be even better.
I say that because I'm a guy who used to bring all his planes home in a 5 gallon bucket.  Robust is good when you are digging them out of the ground.

jjz015, 2022-10-24:
Yup, I lowered the line width since that seemed to decrease the long strings across the part. My plan was to increase the flow rate manually to compensate for it. I will aim for 0.6mm line width tho, I don't wanna have a wing snap on me mid flight.

GregValiant, 2022-10-24:
In spiralize they aren't "strings" but rather planned extrusions that connect those 9 separate areas.  If you had 9 separate calibration cubes and tried to print them all at once in Spiralize you would get the same thing.  One path that goes everywhere while extruding all the way.

GregValiant, 2022-10-26:
I'll close this one in favor of the previous report.

closed 2022-10-26 by GregValiant