Ultimaker/Cura #13633: update or use marketplace without crashing

AlexFaber18, 2022-10-23:
### Application Version

5.2.1

### Platform

windows 10

### Printer

ender 3 

### Reproduction steps

1. downloaded the newest update like version 5.2.0 advised 
2. try the new version out but it wouldn't let me update after I used the marketplace from the main website to add the cutter feature 
3. tried to redownload the previous version and now it has the same problem 
4. both versions crash after I try to update my profile 

### Actual results

i exit the application and think about downloading a different application cause I shouldn't have something like this happening 

### Expected results

my profile should have been updated without the whole application failing 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

n/a

MariMakes, 2022-10-25:
Hey @AlexFaber18,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report, I'm sorry to hear you are stuck 😖 

In this case, we need your configuration folder and your logs for troubleshooting. 

You can share your configuration folder by:
	1. Go to Help at the top bar of Cura
	2. Select Show Configuration Folder
	3. Select all the folders in there and create a zip OR go up a folder in your folder structure and zip your folder with the correct Cura version
	
You can find your log file here:
Windows: %APPDATA%\cura\<Cura version>\cura.log or usually C:\Users\\<your username>\AppData\Roaming\cura\<Cura version>\cura.log

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser. 

Thank you 🙏

AlexFaber18, 2022-10-25:
[5.2.zip](https://github.com/Ultimaker/Cura/files/9859638/5.2.zip)
Thank you for getting back to me so quickly attached above I believe is what you have asked for. Thank you again.

MariMakes, 2022-10-25:
Hey @AlexFaber18,

Thanks for the files.
Unfortunetly, I'm still having trouble recreating your issue, even with your configuration folder. 😖 

I'll have someone look at your included log files. 

Can you share what kind of cutter feature you wanted to add? 
Is it a plugin on the marketplace?

AlexFaber18, 2022-10-25:
So the additional item I want to add is the mesh tool that is able to cut up a stl file that is to big to fit in my ender 3s parameters.

MariMakes, 2022-10-26:
Do you remember what the tool is called? 
Is it something I could install aswell?

AlexFaber18, 2022-10-26:
Yes it is the mesh tool in the marketplace.

MariMakes, 2022-10-26:
[image]

I installed the plug-in but it only has a split model into parts setting. 

But that only works for STLs that consist out of multiple small submeshes like this model: 
[image]

Is that what you want to do?

AlexFaber18, 2022-10-26:
Yes it is that is what I was trying to accomplish

[image]<https://nam12.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F40423138%2F198017916-814cdcf3-f579-4f5f-a80a-372549df145f.png&data=05%7C01%7C%7Cf6825d27641a4812d6bc08dab747e187%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C638023816591319626%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=lZr%2BUqIOy9htKRkbBz6EKCPoDIJcpBztTGxSg5eNt3E%3D&reserved=0>

[image]<https://nam12.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F40423138%2F198018143-f11bf8b0-155e-40ad-8d3b-db16495b44b3.png&data=05%7C01%7C%7Cf6825d27641a4812d6bc08dab747e187%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C638023816591319626%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=QQ6sAifl4honBrKDLnOeOhYqeZZe7vGwNlYje3Se7%2Bc%3D&reserved=0>

AlexFaber18, 2022-10-28:
> It's a bit hard for me to troubleshoot your crashing marketplace. 😖
>
> But I think I can help you get unstuck. You can download the Meshtools plugin from this link.
> https://github.com/fieldOfView/Cura-MeshTools/releases/tag/v3.7.0<https://nam12.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2FfieldOfView%2FCura-MeshTools%2Freleases%2Ftag%2Fv3.7.0&data=05%7C01%7C%7C878fe92b78114f55cd4a08dab8211518%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C638024749473776517%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=njWyMQ0njzwk4Avncx14VdsebN43NzKxVKBtu4QDzmE%3D&reserved=0>
>
> Make sure to download the version that mentions Cura5.0.
> You can download the .curapackage and drag it into your Cura to get it installed.
>
> Good luck. 💪
>
> Can you let us know if you manage to get it installed?

Hello thank you again so much I was able to get to work the way I wanted to. So I really appreciate it. Have a wonderful weekend.

MariMakes, 2022-10-31:
Yay! I'm happy to hear this update. 😊 
I'll close the issue as resolved.

closed 2022-10-31 by MariMakes

AlexFaber18, 2022-10-31:
Thank you again for all your work it was much appreciated

Closed #13633<https://nam12.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2FUltimaker%2FCura%2Fissues%2F13633&data=05%7C01%7C%7Ce5596b1480e34e9b100608dabb22d61b%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C638028055544438264%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=8B24TQD96PNyq9ieHW5Q4m3hxltD%2FlXZt9usdIw%2FfcY%3D&reserved=0> as completed.