Ultimaker/Cura #13634: Camera rotation and zoom slowly break down

Ohmarinus, 2022-10-23:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Custom Replicator 2x

### Reproduction steps

Zooming in, selecting objects, zooming out, etc. Moving around the camera, and eventually I'm not able to zoom in nor adequately pivot the camera around. It progressively gets worse.

The Cura camera rotation pivot will change depending on the zoom level. 

With 'Center camera when item is selected' on, each time you select an item, the camera will move further away, this will cause the problem of not being able to zoom in any further.

The only way to get out of this loop is to restart Cura.

Video here:
https://www.youtube.com/watch?v=y-o2P6qdg28

And another video here:
https://user-images.githubusercontent.com/5002107/197389066-f1f8af1a-26e6-4f05-bab8-80d5b0c2367d.mp4

### Actual results

The zoom and camera rotation seems to degrade over time to the point where I cannot zoom in anymore. There is no way to get out of this zoom level and you end up stuck seeing the bed far away, only zooming in slightly between 'far away' and 'very far away'.

### Expected results

I expect the pivot to either be at the center of the build platform, or on my mouse pointer when I **right-click + drag**.

I expect the zoom level to keep working, being able to zoom onto and even through a model, and also being able to back off again.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

-

MariMakes, 2022-11-08:
Hey @Ohmarinus,

Thanks for your report 👍
Sorry, it took us a while to get back to you 😞 

This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/13514
We have a number of reports on this issue so I'll bring it up to the team to see if we can improve this behavior. 
I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/13514

closed 2022-11-08 by MariMakes

MariMakes, 2023-01-03:
Hey @Ohmarinus,

Just wanted to give you a quick update on this case.
We've had reports that in some cases the Camera Plug-in that is bundled seems to be disabled in some cases. 

Can you doublecheck if your camera is enabled? 
You can go to the Marketplace, click on the Gear and scroll down. 
[image]

Ohmarinus, 2023-01-03:
Hi Mari, just to confirm, It is enabled for me:
[image]

But I stopped using Cura because for some unknown reason the heated bed settings suddenly started to disappear while they are enabled in every other part of the expert settings. Apparently there was no way to get this back working the way it did before.

I can still troubleshoot if you have a question about this issue.