Ultimaker/Cura #13635: crash 5.2.0 and 5.2.1

sankafola, 2022-10-23:
### Application Version

5.2.0

### Platform

Imac 10.15.7

### Printer

Predator Anyquibic 

### Reproduction steps

1-install 5.2 ver or ver 5.2.1 
2-start it 

### Actual results

it does not start even once 

### Expected results

to open the program 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

i tried to install the 5.2.0 and the 5.2.1 and both they dont start for the first time , it just crash and there is no log file created

2x38, 2022-10-23:
Same Error:
Ultimaker Cura 5.1.1 and 5.2.5
Imac core i5 8Go / NVIDIA GeForce GT 640M 512 Mo
MacOS Catalina 10.15.7

MariMakes, 2022-10-25:
Hey @sankafola and @2x38,

Thanks for your report. 👍 
Unfortunately, we no longer support Catalina 10.15, since the services we use, including Apple, are no longer supporting that OS.  [In the release notes you can find a mention to this page](https://support.ultimaker.com/hc/en-us/articles/360011889579), it mentions that the last Cura that is supported for that OS is 5.1.0

I'll be closing this issue since it's not an issue in Cura. I hope you understand 🙏

closed 2022-10-25 by MariMakes