Ultimaker/Cura #13636: Slicing error

iherbak, 2022-10-23:
### Application Version

5.2.1

### Platform

Windows 11

### Printer

TwoTrees SP5

### Reproduction steps

Imported the model scaled to 200%.
Tried to slice

### Actual results

Slicing always failed with issue

### Expected results

Slicing is complete so i could print it

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[TWOTSP_MAKIES_Jack_O_Lantern_Body.zip](https://github.com/Ultimaker/Cura/files/9846894/TWOTSP_MAKIES_Jack_O_Lantern_Body.zip)
[cura.zip](https://github.com/Ultimaker/Cura/files/9846903/cura.zip)

GregValiant, 2022-10-24:
Thanks for the report.
I can duplicate this using your project file.
A workaround is to rotate the model on the build plate.  15° worked for me.
[image]

@Piezoid is this something like we were seeing with 5.1?

gelse, 2022-10-24:
i can confirm that bug, have the same issue (and the same workaround) with the attached project file.
[MSMV2E3D_d20_7mm.zip](https://github.com/Ultimaker/Cura/files/9849801/MSMV2E3D_d20_7mm.zip)

jellespijker, 2024-01-14:
Thank you for your report. It seems that your uploaded project file slices correct on our latest unreleased 5.7.0-alpha.0+3abb5 which indicates that the issue you're experiencing seems to be fixed for the upcoming 5.7.0 release.
I'm closing this issue as fixed for the 5.7.0 release

[image]

closed 2024-01-14 by jellespijker