Ultimaker/Cura #13638: (4.11.0)Remove All Holes feature not working with Support Blocker

M3talB3nder, 2022-10-24:
### Application Version

4.11.0

### Platform

Windows 10

### Printer

Ender 3 pro

### Reproduction steps

1. Noticed unwanted holes in sliced file
2. I used the "Remove all holes" tab and sliced successfully fixing the holes
3. I then added a custom support
4. "Remove all holes" no longer being applied to slice even with check box checked
5. Removed custom supports and used Generate supports which corrected the Remove all holes issue
6. I then used a support blocker and had the same issue happen not allowing the "Remove all holes" tab to work

### Actual results

Remove all holes tab will not work with custom supports or support blockers

### Expected results

Removed holes should still remove holes even with custom supports or support blockers

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

attatched are the project files , one with a support blocker and one without 
[ImageToStl.com_helmet.2.zip](https://github.com/Ultimaker/Cura/files/9848697/ImageToStl.com_helmet.2.zip)

GregValiant, 2022-10-24:
Thanks for the report.
When I open your project file "with support blocker" I get this view.  Do you see anything odd about it?
[image]

Cura is calling for support on the top of the model and for the inside where the floor is.  That indicates errors in the model.

This is the view after repairing the model.  You can see that the red areas are gone from the top side.  That may have been why you noticed "holes" in the model.
[image]

So far this looks like a problem with the model.  If you would repair the model (I used MS 3D Builder included with Windows) and then try again you should get a better result.
Let us know how you make out.

Back to the blocker...
Horizontal Holes are not affected by "Remove All Holes".
In the image below there is a Support Blocker in the double-wall area and it is configured in the Per Model settings to Modify Settings for Overlaps and as a Cutting Mesh with Remove All Holes enabled.  The hole that the blocker covers is gone.

I can't reproduce the problem you were having with the blocker and so I will remove the bug label but leave this open and you can respond here if you feel it should remain open.
[image]

M3talB3nder, 2022-10-24:
You're definitely correct that the model isn't quite right and that may be connected to my issue but this was the issue I was having:

The image below has the bottom ring fully connected to the helmet, this is the file with out custom supports or support blockers and remove all holes is checked and at the very least is doing something. 

[image]

After adding a support blocker as shown in the project file above it almost appears that the even though the remove all holes is still check it is now being ignored and the ring is no longer attached to the helmet. 

[image]

I would have thought that even with the stl being a little quirky that because it sliced correctly with out the support blocker adding one should not have effected the remove all holes feature.  Both pictures were taken at the same layer height. The only thing different between the two pictures was adding the support blocker and the second picture looks the same as if Remove all holes was left unchecked.

GregValiant, 2022-10-25:
Remove all Holes has some very specific restrictions like not working on horizontal holes.
Consider an empty drinking glass.  Remove all holes will fill it to the rim.  If you tip the glass at a 45° angle then one edge goes down and water spills out because the rim is lower on one side.  Below is a slice where I've tipped one 45° as an example.  A cross-hole would do the same sort of thing because vertically, it's a "C" shape and not an "O" shaped feature.
[image]

My point here is that Remove All Holes is quirky by itself.  If there are places that are lower than others and would allow the "water" to run out, then at that point Remove All Holes doesn't do as expected because the shape of the feature on that layer has changed from a closed loop, to a "C" shaped slice because it has an open side.

Your model has some errors that are giving you odd results but it is certainly repairable.

MariMakes, 2022-11-10:
Hey @M3talB3nder,

Welcome to the Ultimaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

To me, it doesn't seem like a bug in Ultimaker Cura, since the mesh is faulty and Cura is not built to alter meshes in this way. That is what CAD software is for. 

Apologies for any inconvenience caused by our delayed response. 
I'll be closing this issue. I hope you understand.

closed 2022-11-10 by MariMakes