Ultimaker/Cura #13639: When saving a .gcode file the name no longer pre-populates with the job name

TangoDeltaDelta, 2022-10-24:
### Application Version

5.2.1

### Platform

Linux (not modern)

### Printer

Creality Ender 3

### Reproduction steps

1. Import a model - e.g. "TestPrint.stl"
2. Note the job name populates with "CE3_TestPrint"
3. After previewing, click "Save to Disk"

### Actual results

Note that the file save dialog file name area is empty. 

### Expected results

Previously it would populate with the job name + ".gcode"
e.g. "CE3_TestPrint.gcode" would be pre-loaded into the file name save dialog.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

This I think actually started with the 5.0.0 versions

MariMakes, 2022-10-25:
Hey @TangoDeltaDelta,

Thanks for your report 👍 
I'm having some trouble recreating on my side.

Could it be that you unchecked this setting in your preferences by accident?
[image]

Or did you recently install the [Printjob Naming Plugin](https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/CustomJobPrefix) and altered your preferences there?

fieldOfView, 2022-10-25:
This is a duplicate of https://github.com/Ultimaker/Cura/issues/12659, see my comment here:
https://github.com/Ultimaker/Cura/issues/12659#issuecomment-1173046597

closed 2022-10-25 by fieldOfView

MariMakes, 2023-03-31:
Quick update on our side. 

We might have a fix for this issue. 🎉 
It's a bit hard for us to test, but we have a nightly build available. 
Can you please try it out and let us know if it resolves your issue? 

[You can find the download links and the discussion here. ](https://github.com/Ultimaker/Cura/discussions/15098)