Ultimaker/Cura #13642: Slicing Fehlgeschlagen

kokoalabear, 2022-10-24:
### Application Version

5.1

### Platform

Windows 11

### Printer

Anycubic i3 Mega Pro

### Reproduction steps

2022-10-24 19:52:08,871 - DEBUG - [MainThread] cura.AutoSave._onTimeout [64]: Autosaving preferences, instances and profiles took 0.006005287170410156 seconds
2022-10-24 19:52:30,503 - ERROR - [MainThread] CuraEngineBackend.CuraEngineBackend._onBackendQuit [959]: Backend exited abnormally with return code 3221225477!
2022-10-24 19:52:30,504 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [359]: Attempting to kill the engine process
2022-10-24 19:52:30,505 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [365]: Killing engine process
2022-10-24 19:52:30,505 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [368]: Engine process is killed. Received return code 3221225477
2022-10-24 19:52:30,506 - DEBUG - [MainThread] UM.Backend.Backend._createSocket [231]: Previous socket existed. Closing that first.
2022-10-24 19:52:30,507 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [188]: Socket state changed to Listening
2022-10-24 19:52:30,510 - INFO - [MainThread] UM.Backend.Backend.startEngine [91]: Started engine process: C:\Program Files\Ultimaker Cura 5.1.0\CuraEngine.exe
2022-10-24 19:52:30,511 - DEBUG - [MainThread] UM.Backend.Backend._backendLog [111]: [Backend] Calling engine with: ['C:\\Program Files\\Ultimaker Cura 5.1.0\\CuraEngine.exe', 'connect', '127.0.0.1:49674', '']
2022-10-24 19:52:30,576 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [111]: [Backend] 
2022-10-24 19:52:30,577 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [111]: [Backend] Cura_SteamEngine version main
2022-10-24 19:52:30,577 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [111]: [Backend] Copyright (C) 2022 Ultimaker
2022-10-24 19:52:30,577 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [111]: [Backend] 
2022-10-24 19:52:30,577 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [111]: [Backend] This program is free software: you can redistribute it and/or modify
2022-10-24 19:52:30,578 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [111]: [Backend] it under the terms of the GNU Affero General Public License as published by
2022-10-24 19:52:30,578 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [111]: [Backend] the Free Software Foundation, either version 3 of the License, or
2022-10-24 19:52:30,578 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [111]: [Backend] (at your option) any later version.
2022-10-24 19:52:30,579 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [111]: [Backend] 
2022-10-24 19:52:30,579 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [111]: [Backend] This program is distributed in the hope that it will be useful,
2022-10-24 19:52:30,579 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [111]: [Backend] but WITHOUT ANY WARRANTY; without even the implied warranty of
2022-10-24 19:52:30,579 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [111]: [Backend] MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
2022-10-24 19:52:30,580 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [111]: [Backend] GNU Affero General Public License for more details.
2022-10-24 19:52:30,580 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [111]: [Backend] 
2022-10-24 19:52:30,580 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [111]: [Backend] You should have received a copy of the GNU Affero General Public License
2022-10-24 19:52:30,580 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [111]: [Backend] along with this program.  If not, see <http://www.gnu.org/licenses/>.
2022-10-24 19:52:30,583 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [192]: Socket state changed to Connected
2022-10-24 19:52:30,584 - DEBUG - [MainThread] UM.Backend.Backend._onSocketStateChanged [181]: Backend connected on port 49674

### Actual results

not able to slice

### Expected results

?

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

log adde on top

GregValiant, 2022-10-24:
Thanks for the report.
I had thought this one was fixed.  Maybe it's a Windows 11 thing.
_"Backend exited abnormally with return code 3221225477"_

MariMakes, 2023-02-23:
Hey @kokoalabear,

Thanks for your report 👍

Return code 3221225477, is actually a super generic error, even though it doesn't sound generic 😅
In the Cura 5.3 Beta, we made a fix that should solve a number of these errors.
Can you download the Beta and retry your slice?
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.3.0-beta.1

If that's it still doesn't work, we'll need a little bit more information from you.
Do you have a project file for us? It contains the printer and settings we need for troubleshooting.
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the view it on GitHub link at the bottom of your email to open GitHub in the browser.

closed 2023-03-10 by github-actions[bot]