Ultimaker/Cura #13643: Slicing bug

Zillaccia, 2022-10-24:
### Application Version

5.2.1

### Platform

windows 10

### Printer

FLSUN QQ_S pro

### Reproduction steps

Using the last parametrer used, i launced the slice on the design already suicefully sliced with the v5.2.0, incurring in a bug

### Actual results

a not sliced file

### Expected results

[isabug.zip](https://github.com/Ultimaker/Cura/files/9854503/isabug.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

hi

GregValiant, 2022-10-24:
Thanks for the report.
I can duplicate the problem with your project file.  A workaround is to move and/or rotate the model on the build plate.
Someone from the Cura team will take a look at this.

GregValiant, 2022-10-29:
#13685 is a symmetrical object that won't slice unless rotated 90° and it really shouldn't matter on that one.

MariMakes, 2022-12-01:
Hey @Zillaccia,

Welcome to the Ultimaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

Thanks for your report 👍
I can recreate your issue on my side:
[image]

I've noticed that if you move the model slightly from the center of the buildplate, you are able to slice it. 
I hope that helps you get unstuck. 
[image]

I've also started collecting issues that have the same "workaround" and I'll bring it up to the team to see if it can be resolved.
Fingers crossed that they can find something 🤞 

Thanks again! 💪

MariMakes, 2022-12-05:
We've added a ticket to the backlog with the intent to improve this. 
For internal reference CURA-9976

Thanks for the report! 👍

closed 2023-01-27 by casperlamboo