Ultimaker/Cura #13644: z-hop is not working

guate6, 2022-10-24:
### Application Version

5.2.1 and 5.2.0

### Platform

windows 11

### Printer

ultimaker s5

### Reproduction steps

Load up a dual material print with prime tower

### Actual results

Although z-hop is enabled, the machine does not z-hop

### Expected results

It should z-hop

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

https://www.dropbox.com/s/hg2a5t6qsk38l31/Woodfill%20test%20object%20-%20boat.3mf?dl=0

The test file

GregValiant, 2022-10-24:
Thanks for the report.
I sliced the file and generated a gcode.  I made a couple changes to make the gcode easier to search.  (I turned on Firmware retraction and changed the z-hop speed to 11.1).
There were 3059 retractions and 360 Zhops.

I notice that you have "Max Comb Distance with No Retract" set to "0" which means no retracts for combing moves.
Changing some numbers to increase the number of retractions resulting in:
5910 retractions and 360 Zhops
That looks like there is something going on there.

I switched to a dual extruder Ender 3 Pro and these numbers came up:
5253 retractions and 2927 Zhops.

That makes it appear that the problem is in the settings or profile for the S5.

Someone from the Cura team will take a look.

wulfhednar02, 2024-01-28:
Whenever "Z Hop when retracted" is enabled, "Z Hop only over printed parts" is disabled, and "Z Hop Height is set to 0.5mm that the Z Hop would not work / get applied? I seem to be having problems with it not working in the current version, 5.6.0. I have not found a rhyme or reason to it yet, as of now it just works most of the time and sometimes it does not work.

Therefore, I am asking to try and figure out what it might be.

wulfhednar02, 2024-02-04:
@GregValiant I have a print that I cannot get to work with Z Hop. I have tried several different settings, and I am at 4 failed prints now. I got the adhesion to be good enough for it finally, and now the stepper motor is skipping occasionally and layer offsets happened. I looked through the GCODE and there were no Z hops after retractions, and retractions are working just fine. Cura version is 5.6.0.

GregValiant, 2024-02-04:
@wulfhednar02  Zip a project file and post the zip file and I'll take a look.  Using my settings and printer I can't duplicate the problem.  That doesn't mean there isn't a problem.

wulfhednar02, 2024-02-05:
[handle.zip](https://github.com/Ultimaker/Cura/files/14167152/handle.zip)

I have found that with my settings, the retraction works fine in both settings, but the Z Hop only shows up if I turn off combing. With any combing, the Z Hop no longer happens during retractions.