Ultimaker/Cura #13645: Slicing Error

mardoudan, 2022-10-25:
### Application Version

5.2.1

### Platform

Apple MAC OS 12.6

### Printer

CR10SPROV2

### Reproduction steps

1. Opened a CURA 5.2.1 Project
2. Slected Slice
3. CURA Crashed
4. Can't Slice Model
[mando_full.3mf.zip](https://github.com/Ultimaker/Cura/files/9856485/mando_full.3mf.zip)

### Actual results

[mando_full.3mf.zip](https://github.com/Ultimaker/Cura/files/9856506/mando_full.3mf.zip)

### Expected results

Model should slice

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[mando_full.3mf.zip](https://github.com/Ultimaker/Cura/files/9856509/mando_full.3mf.zip)

MariMakes, 2022-10-25:
Hey @mardoudan ,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report. 👍 

I took a look at the file you shared but i've been able to slice it in Cura 5.2.1 on my windows device. 
[image]

Are you sure you shared the correct file? 
Have you tried slighlty moving the model away from the center of the buildplate? 
That causes Cura to recheck the settings and sometimes enables slicing where it couldn't before.

mardoudan, 2022-10-25:
Those are great suggestions. I deleted the support block and was able to slice it.  I’ll put the support block back and try to Slice again using your suggestions. Many thanks, Mark D’Andrea

MariMakes, 2022-10-25:
Hey @mardoudan,

Does that mean you are still stuck? 
Or can we close the issue?

mardoudan, 2022-10-25:
No - I’ll remember your work around- thank you 😊 Regards,Mark D’Andrea

MariMakes, 2022-10-26:
In that case, I'll close the issue... 
Happy printing! It looks like a really cool and rewarding project. 💪

closed 2022-10-26 by MariMakes