Ultimaker/Cura #13646: Slicing failed with an unexpected error - Cura 5.2.1

RowBoCop, 2022-10-25:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Ender 3 S1 Plus

### Reproduction steps

Tried slicing in 5.1.0 and 5.2.0 with same error
Tried slicing with different profile with same error
Tried to repair model in Meshmixer, same error

### Actual results

Error 'Slicing failed with an unexpected error'

### Expected results

Expected successful slice, have sliced two files from the same set of models without issue using same profile and version

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Project file too large to upload here using the required file type, uploaded here - https://drive.google.com/file/d/1Epj6Ddx36L2O4UQghRXW1ptCxftk8G1e/view?usp=sharing

MariMakes, 2022-10-25:
Hey @RowBoCop,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report. 

I can recreate your issue on my side, so I'll have someone from the team look at it. 
[image]

GregValiant, 2022-10-25:
Hello @MariMakes and @RowBoCop ,
This is another extremely high-resolution model.  It's over 1.1 million vertices and over 2.3 million triangles.  Mesh Mixer reported no errors in the model.

After exporting the model as an STL and doing a level 1 simplification in MS 3D Builder the model did slice.  It's an 88 hour print from a 10.3 million line gcode file that is 336mb in size.

I've been unable to get it to slice a 2nd time.

RowBoCop, 2022-10-25:
Thanks! I will see if I can get it to slice after simplifying the model

closed 2022-10-25 by RowBoCop