Ultimaker/Cura #13647: Slicing error in Cura for big model (5.1.0)

heinrichWeichert, 2022-10-25:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Anycubic I3 Mega S

### Reproduction steps

Sliced model

### Actual results

Failure with error popup

### Expected results

No failure

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9858397/cura.log)

https://we.tl/t-bZOg6XHeJd

The error does not occur if I slice with scaling set to 50%, 90% and other values < 100%

MariMakes, 2022-10-25:
Hey @heinrichWeichert,

Welcome to the Ultimaker Cura Github 🚀 

That's a cool model! But this mesh seems faulty 😖 
Have you tried doing a quick mesh fix with the Meshtools Plugin? 
You can get it here: https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/MeshTools
If that doesn't fix it you can try a more elaborate mesh fixer like https://myminifactory.github.io/Fast-Quadric-Mesh-Simplification/.

[image]

heinrichWeichert, 2022-10-25:
Thanks for the quick reply, the mesh fix seemed to help.

closed 2022-10-25 by heinrichWeichert