Ultimaker/Cura #13650: Unable to open a .3mf project file

Tom4557, 2022-10-25:
### Application Version

5.2.1

### Platform

Windows 11 Pro

### Printer

CR-10S Pro V2

### Reproduction steps

Installed Cura 5.2.1 and performed a Method 2 for a clean install following link, https://support.ultimaker.com/hc/en-us/articles/360013007080.

When I Open a .3mf file I get a blank Open Project screen that does nothing.

[Cura errors.docx](https://github.com/Ultimaker/Cura/files/9859981/Cura.errors.docx)

### Actual results

Still the same result

### Expected results

A file that is open on the bed of the bed in cura.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

Enclosed is the log file 
[cura.log](https://github.com/Ultimaker/Cura/files/9860027/cura.log)

MariMakes, 2022-10-27:
Hey @Tom4557,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍 

Are you seeing this behavior for all project files or a specific one?
I noticed that you are trying to print for a Creality CR-10S Pro V2, but in Cura we have only have a  "Creality CR-10S Pro" available, could it be related to having the correct printer definitons in your Cura version? 

I'll have someone from the team take a look at your logs. 
Maybe they can see what is going wrong. 🤞

Tom4557, 2022-10-27:
Sir,

What I am trying to load is a PLA profile supplied by Chuck Hellebuyck on
youtube.

I can open a .stl file just fine.

MariMakes, 2022-10-28:
Hey @Tom4557,

Ahhh... Do I understand correctly that you are following these instructions from CHEP?
https://thangs.com/designer/CHEP/3d-model/Cura%2520V5.0%2520Best%2520%280.12%29%2520Profile%2520from%2520Filament%2520Friday-184722

The file loads correctly for me, Is this the file that you are experiencing an error with? 
[image]

Tom4557, 2022-10-28:
It was the 5.1 profile not the 5.0 but yes in short.

Tom4557, 2022-10-28:
If I follow the band aid fix for this link, it kinda works.

https://github.com/Ultimaker/Cura/issues/12197

I am guessing that it has something to do with a second monitor!  I do have
two monitors and a nvidia video card.

MariMakes, 2022-10-28:
Oh! That's super nice to hear 😄 
If your issue is resolved, I'll close this ticket. 👍 

Happy printing

closed 2022-10-28 by MariMakes

Tom4557, 2022-10-28:
No, it's not resolved. I don't want to run with one monitor if I have two.
Whatever that service is may be, the key to the whole thing.

MariMakes, 2022-10-28:
But if it's a duplicate we usually close the as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/12197

Is it not a duplicate?

Tom4557, 2022-10-28:
Yes, I guess it would be.

GoguyT3D, 2023-08-01:
> Oh! That's super nice to hear 😄 If your issue is resolved, I'll close this ticket. 👍
> 
> Happy printing

I do not have a dual monitor, although I do have the same problem as @Tom4557 , I try to open a file, it prompts me to open as a project or import, i select project then the blank project screen comes up and does nothing
[image]
This was a problem in 5.4alpha nightly too. I do have a Nvidia Geforce 1050TI graphics card

GoguyT3D, 2023-08-01:
[Uploading New folder.zip…]()
Here is my config folder