Ultimaker/Cura #13653: 5.2.1 hangs when opening dialogs

DKiefer770, 2022-10-25:
### Application Version

5.2.1

### Platform

Windows 11

### Printer

Anycubic MegaX

### Reproduction steps

Attempting to use any function that opens a dialog box causes application to freeze - Create New profile for example.

### Actual results

Application freezes, must end task.

### Expected results

Normal dialog should appear.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9864317/cura.log)

MariMakes, 2022-10-27:
Hey @DKiefer770,

Thanks for your report 👍

I'll have someone from the team look at your logs.

Do you perhaps have a screenshot of what you are seeing? 
We might need your configuration folder to see if any of the profiles, or plugins are making your cura hang. 

You can share your configuration folder by:
	1. Go to Help at the top bar of Cura
	2. Select Show Configuration Folder
	3. Select all the folders in there and create a zip OR go up a folder in your folder structure and zip your folder with the correct Cura version
You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

DKiefer770, 2022-10-27:
[5.2.zip](https://github.com/Ultimaker/Cura/files/9881193/5.2.zip)

MariMakes, 2022-11-01:
Hey @DKiefer770,

I brought it up the team and they'll be looking at your logs.

I was browsing our known issues.
Could your issue be related to this issue: https://github.com/Ultimaker/Cura/issues/12264 ?

MariMakes, 2022-11-01:
Thanks for getting back to us. 👍 

We have a ticket on our backlog with the intent to improve this behavior.
I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/12264

closed 2022-11-01 by MariMakes