Ultimaker/Cura #13657: Missing 3DJAKE from materials

chasrobertson, 2022-10-26:
### Application Version

5.2.1

### Platform

Windows 11

### Printer

WEEDO X40

### Reproduction steps

Many Missing Material Profiles

### Actual results

Require 3DJAKE but only 3 manufacturers visible

### Expected results

Much larger list of material Manufacturers

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

Only 3 Masterial Manufacturers now available

MariMakes, 2022-10-27:
Hey @chasrobertson,

Thanks for your report 👍

Most materials should have a generic version available to be printed with. 
What quality issues are you seeing when select the generic version of the material you want to print with? 

All non Ultimaker printers are contributed by volunteers in the community. When looking at your printer I can see that the original contributor excluded a number of materials on purpose since the profiles were not available. 
https://github.com/Ultimaker/Cura/blob/main/resources/definitions/weedo_x40.def.json

Since Cura is open source, you or anyone else is more than welcome to work on adding these materials and create a pull request yourself here: https://github.com/Ultimaker/fdm_materials/pulls?q=is%3Apr+is%3Aopen
If you are stuck in any way, please reach out! I'll be happy to help out.

chasrobertson, 2022-10-27:
Hi,
Thanks for getting back to me, but I think you may have missed the
point. I all the earlier versions of CURA, the materials folder is
populated with a large number of suppliers including 3DJake. In the
latest version 5.2.1, there are now only 3 material suppliers, and the
rest are missing. I have copied the materials folder from 5.1 and it
works in 5.2.1 as it did in 5.1 with the scrolling error still present
as I reported earlier (it was agreed that it was an error in the
program). Note|: the materials folder is not dependent upon the type of
printer selected and is available to all printers.

Regards

Chas.Robertson.

>>> MariMakes ***@***.***> 27/10/2022 15:16 >>>
Hey @chasrobertson
( https://github.com/chasrobertson) ,
Thanks for your report ��
Most materials should have a generic version available to be printed
with.
What quality issues are you seeing when select the generic version of
the material you want to print with?
All non Ultimaker printers are contributed by volunteers in the
community. When looking at your printer I can see that the original
contributor excluded a number of materials on purpose since the profiles
were not available.
https://github.com/Ultimaker/Cura/blob/main/resources/definitions/weedo_x40.def.json
Since Cura is open source, you or anyone else is more than welcome to
work on adding these materials and create a pull request yourself here:
https://github.com/Ultimaker/fdm_materials/pulls?q=is%3Apr+is%3Aopen
If you are stuck in any way, please reach out! I'll be happy to help
out.
—
(
https://github.com/Ultimaker/Cura/issues/13657#issuecomment-1293592689)
(

MariMakes, 2022-10-28:
Hey @chasrobertson,

I can see that you used to have 3D jake in your configuration in https://github.com/Ultimaker/Cura/issues/12851#issuecomment-1194297638
But when I check the materials shipped with Cura, 3D jake is not in the list. 
https://github.com/Ultimaker/fdm_materials
So I don't really know how to help you.
Perhaps there is someone else in this Github who might be able to help you out. 

Could you have gotten the materials somewhere else?
Perhaps downloaded from a website or created them yourself?

closed 2022-11-11 by github-actions[bot]