Ultimaker/Cura #13659: Multiextruder parameters revert to values of Extruder 1 when using "Not overridden"

Block137, 2022-10-26:
### Application Version

5.2.0

### Platform

Windows 10

### Printer

Custom

### Reproduction steps

1. Setup multiextruder printer in Cura 5.x (this problem does not exist in Cura 4.x)
2. Load and assign Extruder 2 to a model
3. Set `Wall Extruder`, `Infill Extruder`, etc to `Not overridden`
4. Change printing parameter for Extruder 2 such as wall count and speed
5. Click anywhere outside the field to complete change

### Actual results

Certain parameter fields revert back to previous value after an attempt to change which are the values of Extruder 1.
Changing values for Extruder 1 change them for all extruders.
Using specific extruder instead of `Not overridden` allows setting different values for each extruder again but this make multiextruder pointless.

### Expected results

Walls, top-bottom, Infill, Speed and other parameters of each extruder without chain icon are independent.
(same behavior as in Cura 4.x)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9869514/cura.log)
[project.zip](https://github.com/Ultimaker/Cura/files/9869516/project.zip)

MariMakes, 2022-10-27:
Hey @Block137,

Thanks for your report 👍
That's a beautiful model ❤️ 
I can reproduce your isue, that definetly looks like a bug. 🐛 

I did notice that when I disabled the first extruder the expected behavior shows up.
So I if that is an option for you, I hope that helps you get unstuck.

https://user-images.githubusercontent.com/40423138/198312587-a5cdac09-9a61-46f8-82d5-dbb13ab7ade9.mp4

I'll bring up your issue to the team to see if they can do something about it. 🤞

Block137, 2022-11-16:
Found more possibly related bug with line width.

The line width for extruder 2 is set to 0.45mm and 0.285mm for extruder 1.
The model sliced with different width between brim and model, presumably using 0.285mm of extruder 1 on the model and 0.45mm for brim. Similarly, disabling extruder 1 fixed it but makes extruder 1 unusable at the same time.
[image]

Project file: 
[LineWidth.zip](https://github.com/Ultimaker/Cura/files/10021192/LineWidth.zip)

MariMakes, 2023-03-13:
Hey @Block137m

Apologies for the lack of updates on our side. 

Someone in the team suggested trying again with the “tab” or Enter action as a workaround.
But that doesn't really seem to work for me, perhaps it works for you. 

Since we only have 50 slices per day with >3 extruders vs 500,000 slices with 2 or less extruders we've decided to defer the issue to be able to focus on the most requested and needed features and bug fixes from our community and from UltiMaker.

Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.
I'll close this issue, I hope you understand.

closed 2023-03-13 by MariMakes

Block137, 2023-03-14:
@MariMakes Are you implying that this bug does not affect 2 extruders profiles? If so, is it possible that this is actually a bug in my printer profile?
Can you replicate this bug with newly created test profile?