Ultimaker/Cura #13662: Concentric ironing issue

jasonwhitlock, 2022-10-26:
### Application Version

5.2.1

### Platform

Mac OS

### Printer

FLsun QQ-s Pro

### Reproduction steps

Using a Normal Profile as a base, I made adjustments that I normally do to a file. Then I used Ironing ZigZag, it sliced and printed fine. Just for fun I changed the ironing to Concentric and I got an error every time.
[image]

### Actual results

Slicing error. 

### Expected results

The Model should have sliced and I would have the G-code.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

.

jasonwhitlock, 2022-10-26:
I Could not upload the project file so I put it in my dropbox: https://www.dropbox.com/s/jluuc64lmt1wgqo/Concentric%20SNAKE.3mf?dl=0

GregValiant, 2022-10-26:
Thanks for the report.
I check all models for errors before doing anything else.  In the long run it saves time.
This is for your snake:
--> 59 Naked edges (?)
--> 10 Planar holes (?)
--> 0 Non-planar holes (?)
--> 7 Non-manifold edges (?)
--> 19 Inverted faces (?)
--> 0 Degenerate faces (?)
--> 21 Duplicate faces (?)
--> 0 Disjoint shells (?)
That isn't near as bad as the "Crystal Snake" which had around 100,000 errors.  
Loading the repaired model I still had to rotate it on the build plate to get it to slice.  I think this may be a duplicate of a couple of other reports (#13643 is one) where the model wouldn't slice.  This bug was worked on for the 5.2.x releases.
(@Piezoid this may be another.)

[image]

closed 2023-01-27 by casperlamboo