Ultimaker/Cura #13663: bug de la découpe avec remplissage progressif

SONIC-GWADA, 2022-10-26:
### Application Version

5.0.0

### Platform

windows 10

### Printer

SOVOL SV03

### Reproduction steps

Bonjour
Juste utilisé le remplissage progressif. Si le modele est centré j'ai un message de bug, il fait la decoupe si je déplace le modele de quelque mm

### Actual results

message bug de découpe envoyer une notification sur votre site

### Expected results

faire la découpe du model

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9872754/cura.log)
[SS_02_Body.zip](https://github.com/Ultimaker/Cura/files/9872758/SS_02_Body.zip)
[image]

GregValiant, 2022-10-26:
Thanks for the report.
This might be a duplicate of #13643.  Nice job finding the workaround.

MariMakes, 2022-11-08:
Hey @SONIC-GWADA,

Je m'excuse, je ne parle pas français mais je pense que je peux vous aider.

Welcome to the Ultimaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

We fixed a number of slicing errors in Cura 5.2 and when I slice your model in that version of Cura it slices succesfully.

[image]

Can you check in the latest version of Cura and let us know if it's resolved?
Thank you 🙏

SONIC-GWADA, 2022-11-08:
Hello, I have already printed the file by simply moving it one mm. On the other hand I thought that cura would warn of an update which it does not apparently do. So I'm going to get the last update because i get problem with three support

closed 2022-11-08 by SONIC-GWADA

MariMakes, 2022-11-09:
Hey @SONIC-GWADA,

Apologies for any inconvenience caused by our delayed response. 

When a new update is available a pop-up is shown at the bottom of your screen when you open Cura.
If you missed the pop-up you can call it by clicking Extensions | Update Checker | Check for Updates. 
[image]

It could also be that you have this setting unchecked in your preferences:
[image]