Ultimaker/Cura #13664: Ultimaker launches to a blank white window, issue occurs with a single monitor

AustinPhillips2022, 2022-10-26:
### Application Version

5.2.1

### Platform

Windows Server 2019

### Printer

N/A

### Reproduction steps

1. Launch Cura Ultimaker 5.2.1 or 4.7.2, issue occurs whether using the single built in laptop screen or using multiple external monitors 
2. Opens to a blank white window 
[image]

The program is installed on two seperate terminal servers and the issue occurs on both. Initially the issue just occured with 4.7.2, attempted to resolve the issue by updating and the issue persisted.

### Actual results

Ultimaker launches to a blank white window, no toolbar or anything of that nature. Application does not go into a not responding state. 

### Expected results

For the application to launch normally 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9873367/cura.log)

Index303, 2022-10-27:
same，several version occur this issue

MariMakes, 2022-10-27:
Hey @AustinPhillips2022,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

It reminds me of https://github.com/Ultimaker/Cura/issues/8892#issuecomment-1126528037

They fixed their issue by stopping ASUS GPU tweak software that came with their video card.  
Can you try if their workaround works for you? 👼

closed 2022-11-11 by github-actions[bot]