Ultimaker/Cura #13665: Pumpkin sliced with 4.12 slices Ok and prints Ok.  Same pumpkin sliced with 4.13.1 and newer slices, but print fails midway

timradde, 2022-10-26:
### Application Version

4.13.1, 5.20, 5.20.1

### Platform

Mac OSX

### Printer

Creality CR6 SE

### Reproduction steps

1. Slice include STL file will print with 4.12, but not with anything newer.  

### Actual results

Pumpkin prints fine after slicing with 4.12.  Slicing with anything newer has no issues, but print fails midway thru on support.

### Expected results

Would expect this to slice with any Cura and print.  It does not.

There is nothing a log file will show for this.  No errors occur during slicing.  Printing fails and there are
no logs for printers.

This could be a printer issue, but feel it is not.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Pumpkin_1.stl.zip](https://github.com/Ultimaker/Cura/files/9873387/Pumpkin_1.stl.zip)

GregValiant, 2022-10-26:
Thanks for the report.
The model has some problems.  It seems that the 5.x versions of Cura are fussier than previous versions.

--> 80 Naked edges (?)
--> 1 Planar holes (?)
--> 0 Non-planar holes (?)
--> 0 Non-manifold edges (?)
--> 0 Inverted faces (?)
--> 0 Degenerate faces (?)
--> 0 Duplicate faces (?)
--> 0 Disjoint shells (?)
Model errors are often a contributing factor to not slicing.
I can't duplicate this as your model sliced fine for me on my Windows 10 computer.  Someone on the Cura team with a Mac will take a look.  It would be helpful if you would load the model, set Cura up to slice, and use the "File | Save Project" command to create a 3mf file.  Zip the file and post the zip folder here.

timradde, 2022-10-26:
It slices fine in all Curas. It falis to print for all except 4.12. That is the issue.

GregValiant, 2022-10-26:
OK.  Post a 4.12 gcode file that works and a gcode that fails.
What behavior did you note on the failed print?  Did the printhead keep traveling with no extrusion?  Did everything come to a stop?
Are you printing via the SD card?  USB?  Octoprint?

timradde, 2022-10-26:
It seems to be printing support when it goes wonky and then there are strands all over. Like it was trying to span a gap. I will try to include a photo of the last failure. I will attch on next reply as dont see a way to attach here via phone.

timradde, 2022-10-26:
It continues to print, but it's like it missed a few rows in the support
structure.  The support is both in front and in the middle.  The front
is where it happens.  You should be able to see the stranding in the
photo.  Since it appears to slice Ok (even though there may be
minor errors) in all Curas but doesn't print correctly I don't know how to
classify this.  This is a pumpkin from Thingiverse.  If you need
more information let me know.  This may be hard to identify.  I had thought
to compare gcode files, but that's not easy to do.

GregValiant, 2022-10-27:
No, direct comparison is difficult to impossible when one file was sliced in 4.x and the other in 5.x.
There are certain "keys" that can happen to stop a print and that's what I search through them for.  I have macros for both MSExcel and for AutoCad to analyze gcode files.  If the print temperature drops (rare and always a settings issue) or if a command was to be written incorrectly (rare and always an SD card or data transmission error) then a print can fail.  Mostly though I've found these problems to be a printer problem (Sorry for that, but I do keep an open mind.).  Identifying the exact problem can be tough.
If you post gcode files I will go through them.  There could be something in there.

timradde, 2022-10-27:
I understand this is an odd problem. The one time I had a problem like this before was I had been trying to use a 32GB SD card which on the Creality is too large. It would stop extruding but the print head continued to move. I now only use the card that came with it and have had no problems since til this. I can send you the gcode for 4.12, 4.13.1, 5.20, and 5.20.1.

MariMakes, 2022-10-27:
Hey @timradde,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

[image]

I'm getting this error so your mesh seems faulty
Have you tried doing a quick mesh fix with the Meshtools Plugin? 
You can get it here: https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/MeshTools
If that doesn't fix it you can try a more elaborate mesh fixer like https://myminifactory.github.io/Fast-Quadric-Mesh-Simplification/.

If that doesn't fix your issue. 
Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser. 

Some pictures would also really help in this case because a fail midway could also be a hardware issue.

timradde, 2022-10-27:
Is that with version5. I got no such errors. If I had I'd have looked into that. Don't know what to say but dont spend too much time on it. I may just drop back to 4.12 as I have never had an issue with that version.

timradde, 2022-10-27:
What version of Cura would want the project file from or does it matter.

timradde, 2022-10-27:
Yes I have thought this could be a formware issue as its the printing that fails. But Ive print loads of thing and never seen this issue before this.

GregValiant, 2022-10-27:
_"I may just drop back to 4.12 as I have never had an issue with that version."_
That is a definite option.  The 5.x version was extremely ambitious on the part of Ultimaker.  Rolling Arachne in and at the same time moving to Qt6 to accommodate the Mac users may have been a stretch.  Fixing it requires spending time with it and chasing down these sorts of problems in order to figure out the "why".

timradde, 2022-10-27:
I will leave it up to you if you wish to continue to work this or close as not reproducible. I've worked "bugs" myself and know sometimes there is not enough info nor may there ever be. I will to to figure anything on my end. Would prefer to stay on 5 but knowing of this issue think I will go back to 4.12.

MariMakes, 2022-10-28:
Hey @timradde,

It would help if you could share a picture of how the print fails midway. 
It could also be related to under extrusion which could be a hardware issue, not a Cura issue. 

Can you share a picture of how your print job fails?
Thank you 🙏

timradde, 2022-10-28:
Then why does it print just fine when sliced with Cura 4.12.1?  And nothing
newer?  I included the print 904 which shows the
failing on the support in front.  You can still see the holes where the
support hasn't finished or it would be closed on the end.
To me this say for some reason it stopped printing the support and then
many layers later tried to start printing that same
support again, but there was nothing directly underneath to adhere to, thus
the stranding.

MariMakes, 2022-11-01:
Hey @timradde 

I can't see the picture if you include it as an attachment in your email. 
You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser. 

To me, it doesn't seem like a bug in Ultimaker Cura anymore. 
Would you still like to keep this issue open or do you agree that we can close it?

timradde, 2022-11-01:
Just close it. Id like to believe it is not a Cura bug but seems odd it would be a printer bug. I have had the printer for over a year and printed 100s of different thing and this is the only item that failed so drastically. I have dropped back to 4.12.1. I was going to try a different slicer like Prusa but it is enough different that I dont have the time to waste trying to match up params. So just close it as you're not likely to figure it out. I have done my best to try to figure it out myself. I was a programmer for decades so I know what to do to try to resolve a problem but its not easy with to to localize to a specific setting or something else. Thanks for trying.

MariMakes, 2022-11-01:
Thanks for getting back to us. 🎃 
I'll close the ticket.

closed 2022-11-01 by MariMakes

timradde, 2022-11-01:
I would have closed it as not reproducible not completed.  Nothing was
completed.  But so be it.