Ultimaker/Cura #13667: Vase mode starting point for spirelize outer contour not aligned with seam settings

philipphirte, 2022-10-27:
### Application Version

5.1.1

### Platform

MacOS Monterey

### Printer

Ultimaker S5

### Reproduction steps

1. Load model
2. Selected print profile w/ vase mode enabled and custom seam placement e.g. back
3. sliced / looked at preview

### Actual results

[image]

**1 Refers to starting point for vase mode**
The starting point for the vase mode is not adjustable like the z-seam placement
**2 Refers to z seam for bottom layers**

### Expected results

The points of 1 and 2 should be aligned on one line.
Or I need a way to somehow manipulate the placement of the starting point for the "spirelization" phase, because I want it to start at a specific corner.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

**Here is the project file**
https://drive.google.com/file/d/10DD2-1El-HB3D56w-LYeamykzRp74vzp/view?usp=sharing

philipphirte, 2022-10-31:
@GregValiant Which rules does Cura use to decide where the starting point for the vase mode layer after the bottom layers is? Are the start points preferably corners etc.?

philipphirte, 2022-10-31:
[image]
[image]

Still the same problem although the adjustment of the starting position works sometimes with the "same numbers in the travel and custom seam section"-method.

smartavionics, 2022-10-31:
Hi, perhaps I can provide some answers. I put a lot of work into revamping the Cura vase mode some years back. It's probably had some further alterations since then but it should still mostly work how I remember. What it does is ensure that the transition from one layer to another always occurs in a forward direction, i.e. the nozzle should never reverse direction. To satisfy this constraint, the start point of each layer can occur at a different location from where it would be if vase mode is not enabled. The upshot of this is that the seam can wander around as you can see below. I have sliced your project using a temporarily modified Cura that shows where each spiralized layer is starting.

[image]

Incidentally, the Cura layer view isn't really great for vase mode and you do sometimes see artifacts that don't actually  exist when the object is printed, YMMV.

philipphirte, 2022-11-03:
@smartavionics 
Thanks for your answer, but I need to adjust the initial starting point of the continues spiral. 
For example:
- model has 5 bottom layers
- printed in vase mode

Result:
- It will print layers 1-5 normally with a seam which can be seen on the picture I've posted before (referring to the right pink circle)
- After layer 5 it will start to spirelize the outer contour with an initial starting point (referring to the left pink circle on the picture)

**This initial starting point is what I want to change. Which settings impact the location of the initial starting point on layer 5? (i.e. after the bottom layers finished)**

smartavionics, 2022-11-03:
> This initial starting point is what I want to change. Which settings impact the location of the initial starting point on layer 5? (i.e. after the bottom layers finished)

I would have thought that setting the z-seam position like you show above would be sufficient. That does appear to work for my Cura (https://github.com/smartavionics/Cura/releases) which is still based on 4.13 but doesn't appear to work for UM Cura which has moved on greatly since then.

philipphirte, 2022-11-10:
Still not able to influence the position reliably. But I solved it otherwise.

closed 2022-11-10 by philipphirte