Ultimaker/Cura #13668: Profile list obscured by window frame

mindfinder, 2022-10-27:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Ender 3

### Reproduction steps

1. Make sure the Profile list have enough custom profile settings e.g. 15
2. Open 'Print settings' menu
3. Open 'Profile' list drop down menu

### Actual results

'Profiles' drop down extends beyond Cura window, so that bottom list items, like 'Update profile...' and 'Create profile...' are not visible any more

### Expected results

'Profiles' drop down extends up to bottom of Cura window, so that bottom list items, like 'Update profile...' and 'Create profile...' are visible

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

MariMakes, 2022-10-31:
Hey @mindfinder,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

You are not the first person to report this. 
Could you share your resolution settings?
I'm not seeing your issue but my settings look like this: 
[image]

Since it's a known issue, I'll bring it up to the team to see if we can do something about this. 
Fingers crossed 🤞

mindfinder, 2022-10-31:
Hi,

Here are display settings for my 14'' laptop:
[image: image.png]
I didn't even know the scale was set to 150%. I think it was the default
when I installed Windows, as it's even marked as Recommended. Changing
scale to 125% helped:
[image: image.png]

Still Cura could account for scale and adjust drop down to fit the visible
area.

Hope that helps.

Kind regards,
Kris

MariMakes, 2022-11-02:
Hey @mindfinder,

I've discussed your issue with the team.
We've added a ticket to the backlog with the intent to improve this. 
For internal reference CURA-9842

Thanks for the report! 👍