Ultimaker/Cura #13670: Slicing fails with high printing options on Anycubic i3 Mega S

AS-MP, 2022-10-27:
### Application Version

5.2.1

### Platform

Windows 11 pro; AMD Ryzen 7 4700U with Radeon Graphics; 16 GM RAM

### Printer

Anycubic i3Mega S/Pro

### Reproduction steps

1) Start Cura 5.2.1
2) Files > Open File(s) > AIMSPRO_CaTow_Base.3mf
3) Choose option "Open as Project"
4) Keep all Settings in the Summary window and press open

### Actual results

1) The Project is loaded.
2) On slicing the error Message appears:

[image]

**Slicing failed**
Slicing failed with an unexpected error. Please consider reporting a bug on our issue tracker.

3) The Progress bar does not stop working

### Expected results

1) The Project is loaded
2) Slicing works normal and a G-code File is produced

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Project File is too big, I'm sorry. It can be found in WeTransfer:

https://we.tl/t-eItxc9vpCI

Disclaimer: The data of the model belongs to i-lovecraft.com

**Further Informations:**

Slicing does work using normal settings (0.2mm).

GregValiant, 2022-10-27:
Thanks for the report.
I can duplicate the problem.  I rotated and moved the model on the build plate in Cura and it sliced.  I had support turned off and it's a 52 hour print.

AS-MP, 2022-10-27:
Thank you for the answer and your help. 

I only rotated the model on the build plate, turned on the supports and it still sliced. The result is an 58 hour print. 

I think this is a nice work around for thgis problem.

GregValiant, 2022-10-27:
Yes It works and yes it's easy but it really shouldn't be required.  It will get figured out.