Ultimaker/Cura #13673: Slicing failed

galfridus, 2022-10-27:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Ender 3 V2

### Reproduction steps

Printing at .03 line size, tree supports. 

Changed support overhang angle from 33 to 45, sliced.

[High Frontier Holder 3D Stand.zip](https://github.com/Ultimaker/Cura/files/9879481/High.Frontier.Holder.3D.Stand.zip)

### Actual results

Cura shows slicer error message.

### Expected results

Successful slicing.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/9879491/cura.log)

GregValiant, 2022-10-27:
Thanks for the report.
I can duplicate the problem.  Rotating the model got it to slice (it took about 10 minutes to complete).  Sometimes moving the model works as well.
This is a duplicate of several other reports (#13643 is one) and the same workaround is the temporary fix.
The Cura team will take a look.
[image]

galfridus, 2022-10-27:
Thanks for the workaround (changing the angle also fixed it, I was able to slice).

closed 2023-01-27 by casperlamboo