Ultimaker/Cura #13675: UFPWriter: SettingFunction is not JSON serializable

fuzebox1, 2022-10-27:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

custom

### Reproduction steps

Slice
Send to Octoprint

### Actual results

Get error "object of type SettingFunction is not JSON serializable"

### Expected results

Send to Octoprint works as with 5.1.1.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

Apparently this was a bug in the beta of 5.2, but was "fixed". I did not have the beta installed.
At bottom of log you will see an error from UFPWriter, making octoprint addon not able to proceed with write request.

[cura.log](https://github.com/Ultimaker/Cura/files/9881773/cura.log)

MariMakes, 2022-10-28:
Hey @fuzebox1,

Thanks for your report 👍
I'm not sure if this is something that we can solve in Cura. It seems to be related to the Octoprint plug-in. 
That is maintained by our wonderful community member @fieldOfView.

Are you sure you have the latest version?
You can find it here:
https://github.com/fieldOfView/Cura-OctoPrintPlugin/releases

fuzebox1, 2022-10-28:
Hi, I have the latest version, yes.
Fieldofview himself stated it is not a fault in his plugin, but in cura in a discussion thread on the 5.2 beta.
https://community.ultimaker.com/topic/41697-ultimaker-cura-52-released-in-beta/?do=findComment&comment=313331
The issue is said to have been fixed here: https://github.com/Ultimaker/Cura/pull/13473
but i am still getting it.

ckvsoft, 2022-11-04:
Same mistake
in the Linux Modern app image. Versions 5.2.0 and 5.2.1
Version 5.2 Beta2 works. This problem is not with all my printers.

I don't use Octoprint but send it with Connect Moonraker. But as I said in Beta 2 it works with both installed printers.

If I turn off UFP thumbnails it works.

[cura.log](https://github.com/Ultimaker/Cura/files/9942435/cura.log)

MariMakes, 2022-11-07:
Hey @fuzebox1 and @ckvsoft,

I brought your issue up to the team. :flex
They believe it was introduced by this change: https://github.com/Ultimaker/Cura/pull/13209

We've added a ticket to the backlog with the intent to improve this. 
For internal reference CURA-9859

Thanks for the report! 👍

shawnp30, 2022-11-07:
Weird mine only happens with 1 printer. My other printer upload from cura via moonraker with ufp enabled just fine. I can cut off ufp and add script from extensions to add thumbnail to gcode
[image]

Kedryn, 2022-11-27:
I just installed the 5.2.1 from 5.1, and i get this error with my S8 using a profile i made for PETG
i DONT get it with the PLA profile for same printer, so the culprit can be one option in the profile.
I'm trying to figure out which one.

megablue, 2022-11-29:
v5.2.1. i am experiencing this problem as well. it was fine last week.

renatojd, 2022-11-30:
I am getting the same problem, I uninstalled and downloaded the latest version, reinstalled and get the same issue. I am connecting via Moonraker.

jeffoisnz, 2022-11-30:
I am getting this issue, printing using Moonraker as well. Weirdly, it was working fine for numerous prints last night, then this morning I am getting the error. No other changes etc. I have restarted Klipper firmware just to make sure, but not resolved. Once this print is finished (manual gcode upload) I will restart the whole environment and see if that helped.

jeffoisnz, 2022-11-30:
> I am getting this issue, printing using Moonraker as well. Weirdly, it was working fine for numerous prints last night, then this morning I am getting the error. No other changes etc. I have [...]

After rebooting everything, and doing an update on everything to do with Klipper for good measure, I am still getting this issue. Note that inbetween it working and not, I hadn't even closed Cura (I had restarted it now though)

jeffoisnz, 2022-11-30:
> After rebooting everything, and doing an update on everything to do with Klipper for good measure, I am still getting this issue. Note that inbetween it working and not, I hadn't even closed Cura (I [...]

If I disable UFP format, it seems to work (was successfully using UFP previously)

megablue, 2022-12-04:
I think it might have something to do with some recent windows updates, i did a system restore on my windows, reverted back to 18th Nov, I can upload to moonraker without the error again.

inkiiin, 2022-12-04:
Hi @jeffoisnz,
same issue and same behaviour here, 5.2.0b2 works fine.

rochford77, 2022-12-18:
same issue here. Moonraker connection giving the error. Worked great all day then poof. Log attached.
[cura.log](https://github.com/Ultimaker/Cura/files/10253279/cura.log)

rburema, 2022-12-23:
We have some trouble trying to reproduce this. Could any of you maybe send us the profiles it's errored for?

rochford77, 2022-12-23:
> We have some trouble trying to reproduce this. Could any of you maybe send us the profiles it's errored for?

[curaprof.zip](https://github.com/Ultimaker/Cura/files/10296063/curaprof.zip)

sure, enclosed are 2 profiles, broken and working, named accordingly. settings should be "more or less the same"

The plugins I have in cura are:

Moonraker Connectoin 1.6.4 by emtrax
Octoprint Connection 3.7.2 by fieldOfView
Thnigibrowser 3.0.0 by Chris ter bake
XML Material Profiles 1.0.1 by Ultimaker B.V.
Z Offset Setting 3.6.0 by FieldofView

This is in cura 5.2.1 and uploading to moonraker v0.7.1-780-gdde9bcc mainsail v2.4.1 running klipper v0.11.0-22

hope this helps

rburema, 2022-12-23:
@rochford77 Yes, I'm now able to reproduce! Thanks :-)

closed 2022-12-30 by jellespijker

jellespijker, 2022-12-30:
Should be fixed for the 5.3 release

@rochford77 

I noticed that you use some complex logic in your profiles, which were causing additional errors. This happened due to the missing `resolveOrValue('<setting_name>')` call which is needed when the serialization takes place.

This can be fixed by defining values such as:
```
wall_line_count = =1 if magic_spiralize else max(1, round((wall_thickness - wall_line_width_0) / wall_line_width_x) + 1) if wall_thickness != 0 else 0
```

as

```
wall_line_count = =1 if resolveOrValue('magic_spiralize') else max(1, round((resolveOrValue('wall_thickness') - resolveOrValue('wall_line_width_0')) / resolveOrValue('wall_line_width_x')) + 1) if resolveOrValue('wall_thickness') != 0 else 0
```

See the following documentation for a more in-depth explanation: https://github.com/Ultimaker/Cura/blob/main/docs/profiles/getting_a_setting_value.md

ckvsoft, 2022-12-30:
The error also occurs with the Anycubic Chiron profile included in Cura.
Above all, it still worked in version 5.2 beta under Linux

ckvsoft, 2022-12-30:
This profile doesn't look very complex

```{
    "version": 2,
    "name": "Anycubic Chiron",
    "inherits": "fdmprinter",
    "metadata":
    {
        "visible": true,
        "author": "Patrick Glatt",
        "manufacturer": "Anycubic",
        "file_formats": "text/x-gcode",
        "platform": "anycubic_chiron_platform.obj",
        "platform_texture": "anycubic-chiron.png",
        "has_materials": true,
        "preferred_material": "generic_pla",
        "has_machine_quality": true,
        "quality_definition": "anycubic_chiron",
        "preferred_quality_type": "normal",
        "machine_extruder_trains":
        {
            "0": "anycubic_chiron_extruder_0"
        },
        "firmware_file": "MarlinChiron.hex"
    },

    "overrides":
    {
        "machine_name":
        {
            "default_value": "Anycubic Chiron"
        },
        "machine_heated_bed":
        {
            "default_value": true
        },
        "machine_width":
        {
            "default_value": 400
        },
        "machine_height":
        {
            "default_value": 450
        },
        "machine_depth":
        {
            "default_value": 400
        },
        "machine_center_is_zero":
        {
            "default_value": false
        },
        "gantry_height":
        {
            "value": "35"
        },
        "machine_head_with_fans_polygon":
        {
            "default_value":
            [
                [-45, 50],
                [-45, -45],
                [45, 50],
                [45, -45]
            ]
        },
        "machine_gcode_flavor":
        {
            "default_value": "RepRap (Marlin/Sprinter)"
        },
        "machine_start_gcode":
        {
            "default_value": "M107 ;Start with the fan off\nG21 ;Set units to millimeters\nG91 ;Change to relative positioning mode for retract filament and nozzle lifting\nG1 F200 E-3 ;Retract 3mm filament for a clean start\nG92 E0 ;Zero the extruded length\nG1 F1000 Z5 ;Lift the nozzle 5mm before homing axes\nG90 ;Absolute positioning\nM82 ;Set extruder to absolute mode too\nG28 X0 Y0 ;First move X/Y to min endstops\nG28 Z0 ;Then move Z to min endstops\nG1 F1000 Z15 ;After homing lift the nozzle 15mm before start printing\n"
        },
        "machine_end_gcode":
        {
            "default_value": "G91 ;Change to relative positioning mode for filament retraction and nozzle lifting\nG1 F200 E-4;Retract the filament a bit before lifting the nozzle\nG1 F1000 Z5;Lift nozzle 5mm\nG90 ;Change to absolute positioning mode to prepare for part rermoval\nG1 X0 Y400 ;Move the print to max y pos for part rermoval\nM104 S0 ; Turn off hotend\nM106 S0 ; Turn off cooling fan\nM140 S0 ; Turn off bed\nM84 ; Disable motors\n"
        }
    }
}```

jellespijker, 2023-01-01:
The error occurs in the profile which @rochford77 shared https://github.com/Ultimaker/Cura/files/10296063/curaprof.zip the Anycubic Chiron is not a profile but a printer definition. If you are encountering problems it is probably something in the profiles `*.def.cfg`

tispokes, 2023-01-24:
Same here, not in beta 5.0.0
still exists in 5.2.1 after updating plugins

rburema, 2023-01-24:
@tispokes This isn't in the plugins though, you'll have to wait for 5.3-beta at least.

sampascucci, 2023-01-26:
I'm running Cura 5.3.0-alpha+xmas, and I'm receiving this error with custom profiles just like others reported. The exact change I made to trigger the error was to change "Printing Temperature Initial Layer" and save those changes to the profile. The profile became broken after that, and the only way to make it work is to modify "Printing Temperature Layer" to anything else, then it works fine. Even if I change it back to the original value, if it has that blue reset arrow (changes not saved to profile) it works fine. As soon as I hit the blue reset arrow on that field, it breaks again.
[image]
[image]

jellespijker, 2023-01-26:
5.3.0-alpha-xmas is basically a fork of 5.2.1. The fix for this isn't in there. It should be in the 5.3.0-beta release, which comes soon

MariMakes, 2023-01-26:
Hey @sampascucci,

[As mentioned in the discussion:](https://github.com/Ultimaker/Cura/discussions/14111) 
5.3.0 Alpha + Xmas is just a special version of 5.2. The only changes we introduced compared to UltiMaker Cura 5.2.1 are those which are needed for the new supports. So keep in mind, this is not a sneak peek for Cura 5.3 (there are some really cool new features coming up) but a spotlight release highlighting this new version of tree supports.  

So, unfortunately, you'll have to wait for an official 5.3 to see this fix. 
Apologies for the inconvenience.

hybridview, 2023-02-09:
This issue just happened to me suddenly today for the first time on 5.2.1. I also use Moonraker. I tried changing the print temp initial layer as suggested, but it did not work for me. Changing UFP to regular GCODE did fix it though. Error log mentions UFPWriter.
[cura - Copy.log](https://github.com/Ultimaker/Cura/files/10700444/cura.-.Copy.log)

rburema, 2023-02-10:
@hybridview Thanks for the report. As mentioned previously in this thread a number of times, we've fixed this for 5.3, which isn't out yet. The actual 5.3-beta should be out (very?) shortly though!

fieldOfView, 2023-02-10:
>  we've fixed this for 5.3, which isn't out yet.

To further clarify something that has also been brought up before: The fix is not yet part of the Cura 5.3 _alpha_ from last december.

FumbledAgain, 2023-03-01:
I can confirm that this issue still exists in 5.2.1. HOWEVER, there is also an easy workaround because it appears to only occur (in my limited testing) in a specific instance: when the Line Width setting is automatically calculated. If Line Width is entered manually, the problem is resolved. All child settings Wall Line Width, Top/Bottom Line Width, etc.) can all be calculated; only calculating the parent Line Width setting itself causes the problem to be duplicated. Hopefully this helps track this down, if it remains an issue after 5.3.

jellespijker, 2023-03-01:
the 5.3.0 beta is out can you that version

fabiradi, 2023-03-05:
> the 5.3.0 beta is out can you that version

👍🏻 [5.3.0 beta2](https://github.com/Ultimaker/Cura/releases) solves it for me. 

👎🏻 Line width settings did not solve it.

ckvsoft, 2023-03-06:
Beta2 not available, only source

fieldOfView, 2023-03-06:
> Beta2 not available

It is, as a download for Beta1. The downloads ("assets") for beta1 were replaced with beta2, but the title of the release was not changed.

tispokes, 2023-03-15:
> for in 5.3.0 it is fixed but i don't want to update my 5.2.2 version any solution to fix this in 5.2.2 ?

Install Auto backup Plugin, then update with no trouble at all.

MariMakes, 2023-03-15:
Hey @goldjunge91,

You can have two installations next to each other.
Installing a new version doesn't automatically uninstall the previous version. 

Can you share what is preventing you to upgrade to 5.3?

goldjunge91, 2023-03-15:
@MariMakes never change a running system more not :-D  
and I feel all slicers get over the years worst then better of course couple of good futures was implemented but all available slicer right now are not top edge software that runs 100% perfect this is why I don't like to update

ckvsoft, 2023-03-15:
> @MariMakes never change a running system more not :-D and I feel all slicers get over the years worst then better of course couple of good futures was implemented but all available slicer right now are not top edge software that runs 100% perfect this is why I don't like to update

are you a newbie ? If I had never made an update I would probably still be on version 2.
Version 5.2.2 isn't exactly old. above all, it doesn't seem to be running smoothly.

MariMakes, 2023-03-16:
Everyone is free to choose which version of Cura they are running. There is no need for name-calling.

The reason I ask is to understand what hurdles there are that we can take away. Usually, it's things like "IT is stopping me" "I'm worried about losing settings", or "This one bug is completely blocking my workflow". That is something I can help with. 

The feeling that slicers are getting worse is not something I can influence without concrete examples.  

I cannot give you the best of both worlds, but I can point you towards the place where the fix has been applied. 
That's in these two places. 
1. https://github.com/Ultimaker/Cura/pull/14121
2. https://github.com/Ultimaker/Uranium/pull/864

Hope this helps 👍

Hunkoys, 2023-04-12:
Mine had something to do with the profile settings. There are functions in the profiles, like getting the # of perimeters based on the line width and total wall thickness. It seems like when I overridden the auto values, it worked.

PR #14121 (2022-12-30): [CURA-9859] Account for settingfunction value that evaluates to another settingfunction.
Should fix #13675 ... at least our involvement with it. A bit worried about the other errors in that log :-/