Ultimaker/Cura #13676: Slicing failed with an unexpected error

AquaticAcoustician, 2022-10-27:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Creality Ender-3 Pro

### Reproduction steps

1. I loaded the .stl file into Cura, which worked without a problem
2. I selected my preferred profile (also tried it with other profiles, btw)
3. I clicked "Slice"

### Actual results

It does not slice and show this error message: 
[image]

### Expected results

It should slice the model.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[CE3PRO_Arania.zip](https://github.com/Ultimaker/Cura/files/9881833/CE3PRO_Arania.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/9881880/cura.log)

GregValiant, 2022-10-28:
Thanks for the report.
Using the Cura Mesh Tools to check the model this came up:
[image]

This report is from an STL repair site.
--> 0 Naked edges (?)
--> 0 Planar holes (?)
--> 0 Non-planar holes (?)
--> **71 Non-manifold edges** (?)
--> **85 Inverted faces** (?)
--> 0 Degenerate faces (?)
--> 4 Duplicate faces (?)
--> 0 Disjoint shells (?)
Looking at the model in Mesh Mixer most of the errors are around the eyes and mouth.
From a printing standpoint - Here you can see that the face isn't really connected to anything else (and it isn't the only disconnected area).  This might print, but it would fall apart when removed from the build plate.
[image]

The model has errors and although it will print, it won't be a good print.  I don't see a bug here so I'll remove the label.  I'll leave this open if you want to respond or if you feel removing the label was premature.

You could try something like this.  A 2.4mm thick support blocker configured as a cutting mesh with "Remove All Holes" enabled.  The cat would print on a base layer and be part of the model.  (The model still needs repair.)
[image]

MariMakes, 2022-11-10:
Hey @AquaticAcoustician,

Welcome to the Ultimaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

To me, it doesn't seem like a bug in Ultimaker Cura anymore, since the mesh is faulty. 
Would you still like to keep this issue open or do you agree that we can close it?

closed 2022-11-25 by github-actions[bot]