Ultimaker/Cura #13677: Slice barre button

Lucacido, 2022-10-27:
### Is your feature request related to a problem?

from PREVIEW tab
it is difficult to jump from 1 slice level to next level, the lift vertical bar located on the right side, is moving very quickly.

### Describe the solution you'd like

it would be great to have 2 butons to jump to next/previous slice level , 1 level per click.

### Describe alternatives you've considered

to day I have the only solution to use the PLAY button to reach the end of this level, and the bar will jump 1 level when completed.

### Affected users and/or printers

all users using PREVIEW would like it for sure !

### Additional information & file uploads

[image]

MariMakes, 2022-11-01:
Hey @Lucacido,

Thanks for your report 👍

I believe that there is already a way to do this.
If you select the ball you can press your up and down arrow keys to move up and down. 
You can also change the text in the ball to jump to a specific layer. 
https://user-images.githubusercontent.com/40423138/199251963-5abf1a41-ae78-4b77-be34-f741f9821d51.mp4

To me, it doesn't seem like a bug in Ultimaker Cura. 
Would you still like to keep this issue open or do you agree that we can close it?

Lucacido, 2022-11-17:
Hi,

ok, i get the point, you are right, I was looking for a graphic arrow, perhaps the arrow keys are working well.
sorry for my feature request , there is another way to do the job.
thanks for your time

closed 2022-11-17 by Lucacido