Ultimaker/Cura #13679: horizontal hole extension troubble

SeibeGitHub, 2022-10-28:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Anycubic I3 Mega

### Reproduction steps

If I set the horizontal hole extension to -0.15 on the walls, I get individual indentations in the inner walls. See attachment

### Actual results

steps in walls

### Expected results

flat walls

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[Gehaeuse.zip](https://github.com/Ultimaker/Cura/files/9886685/Gehaeuse.zip)

smartavionics, 2022-10-28:
See https://github.com/Ultimaker/Cura/issues/13605.

MariMakes, 2022-10-28:
Hey @SeibeGitHub,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report and thanks for mentioning the duplicate @smartavionics. 👍 

Super interesting case, it's indeed expected behavior but something that would confuse me aswell if I encountered it. 
I'll bring it up to the team to see if we can improve this behavior.  💪