Ultimaker/Cura #13681: "Initial layer print acceleration" can increase accel

richfelker, 2022-10-28:
### Application Version

All

### Platform

All

### Printer

All

### Reproduction steps

1. Setup a profile with high acceleration everywhere but a particular feature (i.e. outer walls).
2. Set an initial layer print acceleration (mandatory if you use "Number of slower layers" nonzero) somewhere below the high acceleration but above the lower acceleration for particular feature(s).

### Actual results

All features of the initial layer print at the fixed "initial layer print acceleration", including the ones set for lower accelerations, and if "number of slower layers" is set greater than 1, that feature in subsequent layers interpolates down from the highr initial layer print acceleration to the requested feature acceleration. This produces ringing and accuracy problems.

### Expected results

"Slowdown" layers should only decrease acceleration, never increase it, for a given feature type. There seems to just be a missing `min()` operation in the logic for computing the initial/interpolated values.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

This problem makes "Number of slower layers" basically unusable without sacrificing quality or ridiculous amounts of print time (using a low initial layer acceleration).

I suspect the exact same issue exists for speed, but I have not checked. It would be quite problematic for different reasons: if you have different line widths for different feature types, a common speed might exceed flow for some of them.

richfelker, 2022-10-28:
Just tested a patch (this is against `CuraEngine` repo) to fix it:

```
diff --git a/src/GCodePathConfig.cpp b/src/GCodePathConfig.cpp
index d96147862..aeb9ab393 100644
--- a/src/GCodePathConfig.cpp
+++ b/src/GCodePathConfig.cpp
@@ -37,9 +37,12 @@ GCodePathConfig::GCodePathConfig(const PrintFeatureType& type, const coord_t lin
 void GCodePathConfig::smoothSpeed(GCodePathConfig::SpeedDerivatives first_layer_config, const LayerIndex& layer_nr, const LayerIndex& max_speed_layer_nr) 
 {
     double max_speed_layer = max_speed_layer_nr;
-    speed_derivatives.speed = (speed_derivatives.speed * layer_nr) / max_speed_layer + (first_layer_config.speed * (max_speed_layer - layer_nr) / max_speed_layer);
-    speed_derivatives.acceleration = (speed_derivatives.acceleration * layer_nr) / max_speed_layer + (first_layer_config.acceleration * (max_speed_layer - layer_nr) / max_speed_layer);
-    speed_derivatives.jerk = (speed_derivatives.jerk * layer_nr) / max_speed_layer + (first_layer_config.jerk * (max_speed_layer - layer_nr) / max_speed_layer);
+    double first_layer_speed = std::min(speed_derivatives.speed, first_layer_config.speed);
+    double first_layer_acceleration = std::min(speed_derivatives.acceleration, first_layer_config.acceleration);
+    double first_layer_jerk = std::min(speed_derivatives.jerk, first_layer_config.jerk);
+    speed_derivatives.speed = (speed_derivatives.speed * layer_nr) / max_speed_layer + (first_layer_speed * (max_speed_layer - layer_nr) / max_speed_layer);
+    speed_derivatives.acceleration = (speed_derivatives.acceleration * layer_nr) / max_speed_layer + (first_layer_acceleration * (max_speed_layer - layer_nr) / max_speed_layer);
+    speed_derivatives.jerk = (speed_derivatives.jerk * layer_nr) / max_speed_layer + (first_layer_jerk * (max_speed_layer - layer_nr) / max_speed_layer);
 }
 
 double GCodePathConfig::getExtrusionMM3perMM() const
```

MariMakes, 2022-11-01:
Hey @richfelker,

Thanks for your report 👍
What a super interesting observation 🔍 
I'll bring it up to the team so they can see if they can resolve this issue.

Thanks 😊

MariMakes, 2022-11-02:
Hey @richfelker,

I brought your suggestion up to the team. 
They see the value of your suggestion and agree that it's a bug. 🎉 

They suggested creating a pull request yourself.
That way we can give you credit for your troubleshooting and research. 

What do you think? 
Would you be open to creating a pull request with your suggested change?

rburema, 2022-11-09:
Fixed by PR of original poster, will be in 5.3 once it passes our QA.

closed 2022-11-09 by rburema