Ultimaker/Cura #13682: Ironing causes slicing to fail

sideswipe2, 2022-10-28:
### Application Version

5.2.0

### Platform

Windows 10 TUF RTX 3080

### Printer

Custom FFF Printer Voxelab Aquila

### Reproduction steps

With ironing enabled the slice took 10 minutes and failed to complete causing an error

with the same settings but ironing disabled slicing completed in a couple seconds

### Actual results

The slicing worked just fine after disabling ironing

### Expected results

Ironing should work

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Ironing Enabled.zip](https://github.com/Ultimaker/Cura/files/9889950/Ironing.Enabled.zip)
[Slicing Completed.zip](https://github.com/Ultimaker/Cura/files/9889951/Slicing.Completed.zip)

MariMakes, 2022-10-31:
Hey @sideswipe2,

Welcome to the Ultimaker Cura Github 🚀 
That's a spicy model you've got there 😉 🍑 

I can recreate your slicing fail on my side as well, so I'll have someone from the team take a look at it. 
I did notice that if you move your model slightly only slightly from the center you can complete the print job.
This model doesn't really need ironing, but I hope it helps to get you unstuck.  
[image]

jellespijker, 2024-01-14:
Thank you for your report. It seems that your uploaded project file slices correct on our latest unreleased 5.7.0-alpha.0+3abb5 which indicates that the issue you're experiencing seems to be fixed for the upcoming 5.7.0 release.
I'm closing this issue as fixed for the 5.7.0 release
[image]

closed 2024-01-14 by jellespijker