Ultimaker/Cura #13686: FLSUN layer teperature FIX

Silenceikillyou, 2022-10-28:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

FLSUN SR

### Reproduction steps

when setting teperature, the printer only responded to first layer teperature and ignored the rest.
WHY: the M104 command to change teprature is in the wrong spot. Its 15 lines too early.
Solution When i manuly moved the command to be just before the layer change and not between movements like it was before it works

### Actual results

printer does not respond to temperature setting as it should

### Expected results

change of temperature after first layer
[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

This should be a quick fix because if i use "modify gcode option" it works so just copy and paste it to do a hotfix

GregValiant, 2022-10-29:
Thanks for the report.
M104 is "Set Hot End Temperature" and it does not matter where it is located in the Gcode file.  When the printer sees the command it should change the hot end temperature.  If you were to send it during a print to tune the print then the Hot End Temperature would change to the new set point.

If you have the Printer Settings plugin loaded you will see settings for "Heat Up Speed" and "Cool Down Speed" and both are in °C/sec.
Cura figures out how long it will take for the hot end to reach the new temperature and when Layer:1 starts the temperature is ready.  Cura uses the M104 command rather than M109 because with M104 the print doesn't pause and wait for the hot end to hit the new temperature setting.  If it did then the nozzle would be in contact with the print and leave a divot.

If the difference between Initial Layer Print Temperature and Print Temperature is small then the M104 line will be right near the Layer:1 line.  If the difference is large then the M104 line goes in earlier so the hot end has time to heat up.
If you change the "Heat Up Speed" to 50°/sec then the M104 line will be just before the Layer:1 line.
If you change the "Heat Up Speed" to 1°/sec then the M104 line will be way before the layer:1 line.
When the printer sees an M104 regardless of where it is within the gcode file. or if it's sent over the USB, or if you use the button on the LCD, the temperature should change.  That's what the command does.

I don't see a bug here as Cura is correctly inserting the M104 command with regards to Heat Up Speed.  Try adjusting the Heat Up Speed and see if it makes a difference.  There may be a glitch in your firmware though.  Temperature changes are not instantaneous but when the printer sees M104 the temperature is supposed to change.
I'm inclined to remove the bug label as this appears to be normal behavior.  Let us know what you think.

Silenceikillyou, 2022-10-29:
Nope FLSUN works in misterious ways and for some reason if i leave it to cura, nothing happens, if i move the M104(or use  modify gcode option which puts it in the right place too) for some reason the printer adjusts teperature immedietly after the first layer is done like it should, that is why i saw it as a bug(maybe the firmware is bugged and doesnt like when the command is in the middle of movements but its much easier to fix it in cura than firmware), i saw people on reddit coplaining about this since i bought my SR and that was some time ago.

GregValiant, 2022-10-29:
Unless the printer/processor is skipping over the command for some reason, it should honor the command and change the temperature.  It absolutely should not matter where it is in the gcode file.  For example:  If you were to raise the temperature for "Wall-Inner" and then drop it back down at the end of the Wall-Inner then it should react to each change.  That is how it's supposed to work.  It isn't any different that an M106 to change the fan speed.  The printer reads it and does it.
What firmware is the printer running?

Silenceikillyou, 2022-10-29:
https://flsun3d.com/pages/sr-firmware version 1.4

GregValiant, 2022-10-29:
Another question - How are you printing...Octoprint?  USB?  SD card?
I did find references to odd behavior with your firmware when printing using Octoprint.
[flsun-qq-s-doesnt-read-temperature](https://community.octoprint.org/t/flsun-qq-s-doesnt-read-temperature/17505)  Refer to  the response by "b-morgan".

Silenceikillyou, 2022-10-29:
I am printing from usb flash disk and before last firmware update i used sdcard which was basicaly the same. Octoprint and usb to pc over serial connection proved unstable at best, so i keep it disconnected.

Silenceikillyou, 2022-10-29:
i would just like to point out that cura is the only slicer that puts the M104 there, prusaslicer for example puts the command where i would expect. Is there a way to fix this by editing come configs in cura or is it more complex?

GregValiant, 2022-10-29:
It would be within the Cura Engine calculations as the actual location of the line in the file is dependent on print speed, length of extrusions, and the other settings I mentioned.  The main usage for all of that is with multi-extruder printers that lower an extruder to "standby temperature" when it isn't being used.  By placing the M104 line earlier in the gcode the extruder has time to heat up to print temp again before it is selected to begin printing again.

This is the first I've heard of the problem but I did find other references to FLSUN printers having the issue.  

The Search and Replace post-processing plugin will accept multiple lines so you could do this to add the temperature change exactly after "LAYER:1":
Search = ;LAYER:1\n
Replace = ;LAYER:1\nM104 S215\n
Use Regular Expressions = True (check the box)

"backslash + n" is the newline character and they are required.

A downside is that Search and Replace doesn't accept the Cura Keywords (you cannot use {material_print_temperature} in the replace box) so if you were to change printing temperature in Cura you will also need to manually change the line in the Replace field.

Silenceikillyou, 2022-10-29:
My dirty fix would be to just use modify gcode option "at layer change" : M104 S215 , and use this instead of the option in print settings. But i was looking for less dirty fix.

GregValiant, 2022-10-29:
Most people would want the gcode adjusted so they know when the set temperature will be reached.  In PrusaSlicer the M104 line is right at the layer change but if you are going from 195 to 210 then it might be 30 seconds and a lot of extrusion while the temperature ramps up.  It's probably 6 of one and 1/2 dozen of the other.  If not for the fact that your firmware seems to be ignoring the command would you really have noticed?
Although I print via the SD card I have my print server software connected to the printer all the time.  During a print the printer sends information back across the USB to keep server software informed as to what is going on.  In order to understand the response, the response needs to be in a specific form (the syntax).  Part of the problem with FLSUN firmware (at least that I've come across) is that the response the firmware makes isn't standard syntax and so isn't understood by a print server.  A standard Marlin response like "echo: busy processing" means the same as "WAIT" but if the print server is looking for the standard Marlin response it won't understand "WAIT".

GregValiant, 2022-10-31:
@Silenceikillyou I'm going to remove the bug label and change this to a discussion and go ahead and close it.  This is expected behavior for Cura and there are settings and other ways to customize it to do what you want.  You can still respond here if you like.

I wonder if FLSUN is aware of these sorts of things.  it seems to have been going on for a while and the problem still exists.

closed 2022-10-31 by GregValiant

Silenceikillyou, 2022-10-31:
@GregValiant OK thank you anyway, at least the is some sort of aswer here how to get aroud the issue if anyone will look for it.