Ultimaker/Cura #13690: NoneType has no attribute replace - Generator MSBuildDeps calling generate() - running Cura from source - Windows 10 - CuraEngine editable in mode

phedg1, 2022-10-29:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Not using a printer

### Reproduction steps

Switch to the latest commits for the 5.2 branches in Cura (latest commit 20 October 2022) and CuraEngine (latest commit 19 October 2022).
Go to the Cura directory and open conandata.yml.
Find the section for "5.2.1", go to the requirements subsection.
Change the entry "curaengine/5.2.1" to "curaengine/5.2.1@phedg1/vanilla".

Open command prompt admin and go to workspace directory. Run the following commands:
powershell

conan editable remove curaengine/5.2.1@phedg1/vanilla
conan remove "*" -f
cd Cura
rmdir venv
cd ..

conan config install https://github.com/ultimaker/conan-config.git
conan profile new default --detect --force

cd CuraEngine
conan editable add . curaengine/5.2.1@phedg1/vanilla
cd ..
cd Cura
conan install . cura/5.2.1@phedg1/vanilla --build=missing --update -o cura:devtools=True -g VirtualPythonEnv

### Actual results

I get the following error in the console:

uranium/5.2.0: Created package revision 9cf9866b81b698746e6edb71c7dc4d3d
conanfile.py (cura/5.2.1@phedg1/vanilla): Generator 'MSBuildDeps' calling 'generate()'
conanfile.py (cura/5.2.1@phedg1/vanilla): ERROR: Traceback (most recent call last):
  File "C:\Program Files\Python\Python310\lib\site-packages\conans\client\generators\__init__.py", line 181, in write_generators
    generator.generate()
  File "C:\Program Files\Python\Python310\lib\site-packages\conan\tools\microsoft\msbuilddeps.py", line 110, in generate
    generator_files = self._content()
  File "C:\Program Files\Python\Python310\lib\site-packages\conan\tools\microsoft\msbuilddeps.py", line 335, in _content
    result.update(self._package_props_files(dep, build=False))
  File "C:\Program Files\Python\Python310\lib\site-packages\conan\tools\microsoft\msbuilddeps.py", line 319, in _package_props_files
    result[vars_filename] = self._vars_props_file(dep, dep_name, cpp_info,
  File "C:\Program Files\Python\Python310\lib\site-packages\conan\tools\microsoft\msbuilddeps.py", line 168, in _vars_props_file
    package_folder = escape_path(dep.package_folder)
  File "C:\Program Files\Python\Python310\lib\site-packages\conan\tools\microsoft\msbuilddeps.py", line 154, in escape_path
    return path.replace("\\", "/").lstrip("/")
AttributeError: 'NoneType' object has no attribute 'replace'

ERROR: Error in generator 'MSBuildDeps': 'NoneType' object has no attribute 'replace'
PS C:\Users\Peter\Documents\3D_Print\_Software\Workspace\Cura>

### Expected results

I expected Cura to build properly.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[Full Log 2022-10-29 Vanilla.txt](https://github.com/Ultimaker/Cura/files/9893272/Full.Log.2022-10-29.Vanilla.txt)

Running Cura from source on Windows 10 on the 5.2 branch, with no requirements in editable mode, works perfectly. Adding curaengine as an editable package in conan causes it to fail. This is running completely vanilla Cura and CuraEngine, with the only change being to the conandata.yml. I only changed this build requirement so I could be extra sure it was using the local one in editable mode.  It is also shown in the log that it is using the right version of Visual Studio (2022 Developer Command Prompt v17.2.5). I do not know how to get it to show the version of MSBuildDeps that is being used.

I have attached a full log of this issue. I am using Python version (3.10.4) and Conan version (1.53.0).

MariMakes, 2022-11-01:
Hey @phedg1,

Thanks for the report. 👍 
Issues related to the developer environment are outside of my comfort zone.

@jellespijker could you take a look, please? 🙏

jellespijker, 2022-11-03:
@phedg1 can you try again? I synced the conan recipes between our `cura` and `cura-ce` artifactory remotes.

maybe first clean your conan cache 
```
conan remove "*" -f
```

phedg1, 2022-11-07:
I have rerun my tests. Including the command "conan remote remove cura" gives the issue of missing mpdecimal.h #13622 . Removing that command from the steps gives the same error message about NoneType having no attribute replace.

phedg1, 2022-11-11:
I have updated visual studio community to 17.4, Conan to 1.54, moved my cura onto the latest commit in the main branch but left curaengine in the 5.2 branch (because the main branch is missing the 5.2.1 version in the conandata file). I removed the cura remote and got the same error.