Ultimaker/Cura #13692: Cura no longer able to send print job to Ultimaker S3 via USB, Cloud, or network!

GoguyT3D, 2022-10-29:
### Application Version

5.2.1

### Platform

Windows 11

### Printer

Ultimaker S3

### Reproduction steps

Tried to print via USB, network or cloud. No success, Reset Digital Factory via U3S and then did a full Printer reset. Cura 5.2.1 claimed some of my configuration files were corrupt and recommended removing all printers and config files so I did. I only added my U3S and the necessary config files for materials and profiles which I had previously exported and tried again, no success. Anything over Cura 5.0 has been bug central for me

### Actual results

Cura creats a crash report and closes

### Expected results

Print

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

This is just after my printer was restored to ultimaker digital factory after the downtime. Full printer reset, digital factory reset, cura wiped all of my printers and config files(all had to be uploaded again)
[machine_instances.zip](https://github.com/Ultimaker/Cura/files/9893673/machine_instances.zip)

GoguyT3D, 2022-10-29:
[image]

GoguyT3D, 2022-10-29:
I have just reverted back to cura 5.0 and two things happened One my ender 3 pro custom printer re appeared without me adding it back, and two: my material profiles from the past (renamed with old wrong name) all appeared and starting syncing with my printer AFTER i already synced the profiles within 5.2.1 now I have duplicate material colors for the same 3rd party filament ie: Black, Black, White, White, Blue, Blue and cura prompted me to "Change printer configuration to Gizmo Dorks TPU 95A Blue" when that is what was selected, so I selected the OTHER Gizmo Dorks TPU95A Blue duplicate and the print started to work. I am going to print this item that is needed this weekend and AGAIN reset my printer, re export my materials in Cura 5.0 and re import them again after my printer reset ( after this print completes) @MariMakes  may know how to interpret this better than anyone else at this point because she has been working with me through this

GoguyT3D, 2022-10-29:
I have reset my printer to default and within cura 5.0 double checked all my materials settings and names in regards to CPE/PETG and all profiles and exported them overwriting the other that were made in cura 5.2.1, Removed my printer from cura, re added my printer to cura, exported my materials with correct names/settings to not trip up cura to the Verbatim USB drive and imported them to my printer successfully, I currently don't have any duplicate profiles on the printer but it seems to happen when materials have the slightest difference in settings causing the same brand to have multiple Red, Red etc. There clearly seems to be some sort of bug with syncing profiles to your printer because of this catastrophe. As of now I am forbidding upgrading to another version of cura because this is the only one which I have had most success. Sorry to burn your ear off @MariMakes

MariMakes, 2022-11-01:
Thanks for your report 👍
Do I understand correctly that you are not stuck anymore? 

I took another look and I've bene able tostart a printjob with your materials in Cura 5.2.1
	1. Go to Help at the top bar of Cura
	2. Select Show Configuration Folder
	3. Open your machine_instances.zip
	4. Drag the contents from your zip to your 5.2 folder
	5. Close your Cura
	6. Open your Cura
	7. Sync your materials in your preferred way. (In my case it was easier to sync them over local network by adding the printer by IP adress)
	8. Slice a printjob 
	9. Select the material under Selecting Other material at the bottom of the list

Here is a video of how I selected the material in the printer:
https://user-images.githubusercontent.com/40423138/199287701-5f415e31-20b8-4a71-a53e-005c895365c9.mp4

If you are running into trouble with syncing materials you can also find instructions on how to use the API to check the files on your printer and delete them individually here: https://support.ultimaker.com/hc/en-us/articles/360016857859-How-to-troubleshoot-issues-with-material-printer-synchronization-

To me, it doesn't seem like a bug in Ultimaker Cura anymore. 
Would you still like to keep this issue open or do you agree that we can close it?

GoguyT3D, 2022-11-01:
It seems that within cura 5.0 with the names and materials I have all my custom profiles and materials properly named only one instance of my 3rd party filament color is shown on my printer (this is after i reset digital factory and did a complete printer reset) though if I upgrade to another higher version of cura old filament names that have since been correctly changed show back up in cura instead of the new ones and cura automatically uploads the old filaments so that my UM3S shows multiple colors in one category when selecting filament on the printer. Honestly I dont know how it does this but the printer would even prompt me to change filament if I selected the wrong Gizmo Dorks CPE Blue out of the two that existed in the list. Currently my printer only has one instance of each color filament because I reset the printer, reset digital factory, uninstalled 5.2.1 reinstalled 5.0 and re imported all my material profiles and filaments manually into cura, sync does seem to work via usb and i was notified that more materials needed to be synced in cura so I guess it works via network too, I can connect to digital factory from my printer, but the only thing that would cause the crashing when selecting custom profiles, multiple instances of a single color of filament on my printer and a U3S printer which is not named as my printer along with my printer to show up in the dropdown menu was upgrading to something above 5.0, honestly I don't think this needs to stay open because I don't plan on updating and wasting another entire weekend trying to completely reset up my printer from scratch to squash this problem agian, as you can see I'm frustrated with all that I had to go through to get this to work, but I guess no problem now because it will be a while before I upgrade cura again. Thank you for your time @MariMakes

closed 2022-11-01 by GoguyT3D